FPGAwars/icestudio #49: Error when verifying/building with an info block

Obijuan, 2016-07-19:
Icestudio version:  0.2.0-beta2-dev
Platform: Ubuntu 16.04

A info block is placed in a blank project. When the verify/build option is pressed, the following error messages appears:   **main.v:2:error: unkwnown module type: main_basic_info**

[image]

[info-test.ice.zip](https://github.com/FPGAwars/icestudio/files/372242/info-test.ice.zip)

Obijuan, 2016-07-19:
If I run **npm install**, a problem with the dependencies appears:

$ npm install

> Icestudio@0.2.0-beta2-dev postinstall /home/obijuan/develop/icestudio
> npm install --prefix app && node_modules/bower/bin/bower install

npm ERR! Linux 4.4.0-31-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v4.4.7
npm ERR! npm  v2.15.8
npm ERR! code EPEERINVALID

npm ERR! peerinvalid The package grunt@0.4.5 does not satisfy its siblings' peerDependencies requirements!
npm ERR! peerinvalid Peer grunt-exec@0.4.7 wants grunt@>=0.4
npm ERR! peerinvalid Peer load-grunt-tasks@3.5.0 wants grunt@>=0.4.0
npm ERR! peerinvalid Peer grunt-contrib-clean@1.0.0 wants grunt@>= 0.4.5
npm ERR! peerinvalid Peer grunt-wiredep@3.0.1 wants grunt@>=0.4.0
npm ERR! peerinvalid Peer grunt-nw-builder@2.0.3 wants grunt@^1.0.1

npm ERR! Please include the following file with any support request:
npm ERR!     /home/obijuan/develop/icestudio/npm-debug.log

Jesus89, 2016-07-19:
Fixed in 4448733c135c49c43de4f5834572fc9913867b6d.

Note: npm install seems to be a problem with grunt: https://github.com/gruntjs/grunt/issues/1488. Current commit prevents this error. Otherwise, try to 'sudo npm uninstall grunt -g'.

Obijuan, 2016-07-20:
The npm problem is not yet fixed. I have tested the latest develop in a clean virtual machine (ubuntu 16.04). I've installed it from scratch following the wiki instructions. The final error can be seen here:

[image]

Right after executing npm install, there appear a lot of warnings:

$ npm install
npm WARN peerDependencies The peer dependency grunt@^1.0.1 included from grunt-nw-builder will no
npm WARN peerDependencies longer be automatically installed to fulfill the peerDependency 
npm WARN peerDependencies in npm 3+. Your application will need to depend on it explicitly.
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated graceful-fs@1.2.3: graceful-fs v3.0.0 and before will fail on node releases >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.
npm WARN deprecated graceful-fs@3.0.8: graceful-fs v3.0.0 and before will fail on node releases >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.
npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated lodash@1.0.2: lodash@<3.0.0 is no longer maintained. Upgrade to lodash@^4.0.0.
\
> nw@0.12.3 postinstall /home/obijuan/develop/icestudio/node_modules/nw
> node scripts/install.js

[image]

Maybe this warnings are related to the problem

Obijuan, 2016-07-20:
Same errors and warnings in 0.2.0-beta1. We should check the warnings and update the dependencies, and see what happens

Obijuan, 2016-07-20:
Fixed! Although some warnings remain:

npm WARN deprecated graceful-fs@3.0.8: graceful-fs v3.0.0 and before will fail on node releases >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated lodash@1.0.2: lodash@<3.0.0 is no longer maintained. Upgrade to lodash@^4.0.0.
npm WARN deprecated graceful-fs@1.2.3: graceful-fs v3.0.0 and before will fail on node releases >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.

closed 2016-07-20 by Obijuan