FPGAwars/icestudio #59: [Feature] Change in the background color of the input-config block

Obijuan, 2016-08-03:
The input-config block is a kind of special block, different than the rest. It only can be used near a input pad and it does not belongs to the digital electronic standard.  So I propose to change the background to a different color, to avoid confusions or error  (Yesterday I made a mistake joining two of this blocks by mistake, because one of them was inside another block. As they had the same color than standard blocks It took me a while to find the error).

After thinking about it, I realized that the config block are different, and should have a different blackground color.

I propose to use a color near to the yellow of the input pads: a kind of light orange or something similar

Jesus89, 2016-08-03:
I prefer to keep two colors. Why not using the same yellow as Input/Output blocks? Because it is related only to these blocks.

Obijuan, 2016-08-03:
Any color different than blue will do the job

Jesus89, 2016-08-03:
Closed 0a92f1ac55a1080e796f0d1183a60737d6b6b51a

closed 2016-08-03 by Jesus89