FPGAwars/icestudio #60: [Feature] Show error message when an input config block is not connected to a input-pin

Obijuan, 2016-08-03:
The input of the input-config block can only be connected to an input-pin. It would be great if there appear an error message if the input-config blocks do not meet this requirement

The message could be something like: "Error: The config block's input does not come from an input pin"

Jesus89, 2016-08-03:
I will notify that error.

Also I will try to add this restriction is wire connection:

* Only input blocks can connect to input-config blocks.

Jesus89, 2016-08-03:
Using two connected config blocks, arachne-pnr says:

`
fatal error: SB_IO PACKAGE_PIN not connected to toplevel port
`

I think a more generic notification must be assigned to this message.

Jesus89, 2016-08-03:
I have added a restriction: https://github.com/Jesus89/icestudio/commit/ed03b13f96af62a5f63654c015a426a5a6bc082f

Two config blocks can not be connected together. I think the notification is not needed.

Obijuan, 2016-08-03:
I think the restriction should be a little bit more general.  Only an input-block is allowed to be conected in the left of an input config block. It includes the case of two input-config blocks connected

Jesus89, 2016-08-03:
Yes, this is the restriction I coded, with better words :)

Jesus89, 2016-08-03:
Closed 0a92f1ac55a1080e796f0d1183a60737d6b6b51a

closed 2016-08-03 by Jesus89