FPGAwars/icestudio #62: Memory file not found

Obijuan, 2016-08-06:
I have implemented a 16x8 rom as a code block. Its content is loaded from an external file. When building the project, it does not find the file.

[image]

The verilog code is (inside the block):

```verilog
reg [7:0] rom [0:15];
reg [7:0] data;
wire [3:0] addr;

//-- Name of the file with the rom contents
parameter ROMFILE = "rom2.list";

assign addr = {a3,a2,a1,a0};

always @(negedge clk) begin
  data <= rom[addr];
end

assign {d7,d6,d5,d4,d3,d2,d1,d0} = data;

initial begin
  $readmemh(ROMFILE, rom);
end
```

This is the icestudio project:

[rom16x8-master.ice.zip](https://github.com/FPGAwars/icestudio/files/405115/rom16x8-master.ice.zip)

Obijuan, 2016-08-06:
For making tests, the memory file can be placed inside the icestudio _build dir.  In that case the file is loaded correctly

Jesus89, 2016-08-06:
External files must be added to "app/_build" directory in order to be detected inside verilog code.

Jesus89, 2016-08-11:
Now .list file is loaded from .ice directory.

closed 2016-08-11 by Jesus89

ke10g, 2020-05-26:
Hi Jesus, sorry to revive this thread but: I am having a problem loading a file. I place the file in my project file's directory but Icestudio does not see it. Any tips?

commit ff0dbed0 (2016-08-11): Merge pull request #63 from Jesus89/develop

Sync list files. Closes #62