FPGAwars/icestudio #65: ACC0 project does not build

Obijuan, 2016-08-12:
I cannot upload the ACC0 project into the Icezum because it does not build. The process to reproduce the problem (in my computer) are:

1) I am using icestudio 2.0 installed from the Released package. Execute it
2) Download the [ACC0.ice](https://github.com/Obijuan/ACC/raw/master/hw/roadmap/10-ACC0/ACC0.ice) and rom.list files
3) Open the project in Icestudio
4) Tools/Build

The Build start notification appears... and after some seconds, it desappears. No success notification is received.

If the project is exported as verilog (and also the .pcf), it is built correctly with apio build (and apio upload). It works ok in the icezum 1.0

I still do not know if this is a bug, or is something related to my computer. I am testing in ubuntu 16.04-64 bits. I need someone to reproduce it

Jesus89, 2016-08-12:
Hi,

This error appears because the stdout of this project is bigger than default nodejs exec maxBuffer:

> maxBuffer <Number> largest amount of data (in bytes) allowed on stdout or stderr - if exceeded child process is killed (Default: 200*1024)

I will update this value to 2000*1024.

closed 2016-08-12 by Jesus89

Obijuan, 2016-08-12:
Great! It works now!!!  :-)  Thanks!

commit 6de5d309 (2016-08-12): Increase exec maxBuffer. Closes #65