FPGAwars/icestudio #93: Is there a plan to support other FPGA than Lattice ?

Martoni, 2016-11-17:
The name of **ice**studio is oriented for ice40. But is there plan to support other platforms ?
I wonder if it's easy to port icestudio for armadeus platforms apf27 (xilinx spartan3a), apf51 (xilinx spartan6) and apf6_sp (Altera/intel CycloneV) ?
Of course, there is no open-source toolchain to synthetize for it, but maybe we can do somethings to drive close-source tools ?

Jesus89, 2016-11-17:
Hi,

We are using **open source tools** for FPGAs (http://www.clifford.at/icestorm/) through apio (https://github.com/FPGAwars/apio). All future Icestorm-supported FPGAs could be integrated.

Icestudio will evolve towards a more hackable editor. Almost everything is possible. Just edit the compiler to generate pcf-like files for specific FPGA and try to generate the bitstream in the console using privative-tools. Other solution could be to use Icestudio only to generate the verilog sources and then import those files to a privative editor.

We won't explore private-tools integration, but maybe another one does.

Martoni, 2016-11-17:
Ok, thanks.

closed 2016-11-17 by Martoni