FPGAwars/icestudio #97: Basic bus specification

Obijuan, 2016-11-21:
It would be great to have parcial implementation of the buses, only for connecting blocks.

Let's take this two blocks as an example, conected by a wire. It is done in icestudio like this:

[image]

If it was a bus (8 bits) instead of a wire, the generated code (in human readable verilog code) will be:

```verilog
`default_nettype none

module main ();

 wire [7:0] w0;  //-- Bus de union

 bloque1 TEST1 (
  .out(w0)
 );

 bloque2 TEST2 (
  .in(w0)
 );
endmodule

//-- Bloque 1: Bus de salida de 8 bits
module bloque1 (output [7:0] out);
 assign out = 8'hFF;

endmodule

//-- Bloque 2: Bus de entrada de 8 bits
module bloque2 (input [7:0] in);
 wire [7:0] temp;

 assign temp = in;

endmodule
```

This is another example: the **connection of 3 blocks**:

[image]

If it was an 8 bits bus, the generated code should be like this:

```verilog

`default_nettype none

module main ();

 wire [7:0] w0;
 wire [7:0] w1;

 assign w1 = w0;

 bloque1 v62df90 (
  .out(w0)
 );

 bloque2 v8cd364 (
  .in(w0)
 );

 bloque3 ve05eb2 (
  .in(w1)
 );

endmodule

module bloque1 (output [7:0] out);
 assign out = 8'hFF;

endmodule

module bloque2 (input [7:0] in);
 wire temp;

 assign temp = in;

endmodule

module bloque3 (input [7:0] in);
 wire temp;

 assign temp = in;
endmodule

```

### Restrictions:

* Only buses with the same width can be connected
* Connection between output buses with input buses (no input/output buses yet)

This implementation will let us to include a lot of new circuit in icestudio

Jesus89, 2016-11-21:
OK. Let's define the generation method.

A simple wire code block with **a**,**b** as input and **c** as output. Is created using `a,b c` syntax. Note the space between `a,b` and `c`.

Which one-line syntax would be interesting for new bus wire code blocks?
- `a[7],b[7] c[7]`
- `a[7:0],b[7:0] c[7:0]`
- `a[0:7],b[0:7] c[0:7]`

Obijuan, 2016-11-22:
I think it is better to use the same syntax than verilog: a[7:0], b[7:0]
c[7:0]

Jesus89, 2016-12-22:
Done: #120

closed 2016-12-22 by Jesus89