FPGAwars/icestudio #108: Enable Drivers Hangs or Requires Internet Connection

r-ohare, 2016-11-28:
On OS X 10.12.1 with Icestudio 0.2.3 Beta, enabling drivers caused a complaint that an internet connection is required. I don't have any internet connectivity problems. After reading the install instructions again I installed the most current Python 2.7. Now enabling drivers causes the app to hang.

Jesus89, 2016-11-28:
Drivers > Enable option in mac installs libftdi using homebrew and also performs some kextuload: http://apiodoc.readthedocs.io/en/stable/source/installation.html#manually. This behavior is equivalent in the cli "apio": https://github.com/FPGAwars/apio.

Can you install it (pip install -U apio), execute this command (apio drivers --enable), and paste the output?

Thanks.

r-ohare, 2016-11-29:
I appear to have had some undone steps. First I had to install pip (even though I already had a functioning python installation) using the get-pip.py script. Then I installed apio using pip. Then I could run the above command. The output follows:

System:~ home$ apio drivers --enable
Configure FTDI drivers for FPGA
/System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/lib/ruby/2.0.0/universal-darwin16/rbconfig.rb:213: warning: Insecure world writable dir /usr/local in PATH, mode 040777
==> Downloading https://homebrew.bintray.com/bottles/libftdi-1.3.sierra.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libftdi-1.3.sierra.bottle.tar.gz
==> Caveats
Python modules have been installed and Homebrew's site-packages is not
in your Python sys.path, so you will not be able to import the modules
this formula installed. If you plan to develop with these modules,
please run:
  mkdir -p /Users/home/Library/Python/2.7/lib/python/site-packages
  echo 'import site; site.addsitedir("/usr/local/lib/python2.7/site-packages")' >> /Users/home/Library/Python/2.7/lib/python/site-packages/homebrew.pth
==> Summary
🍺  /usr/local/Cellar/libftdi/1.3: 23 files, 419.7K
Password:
FPGA drivers enabled
System:~ home$ 

After having done the above, Icestudio can enable and disable the drivers without any difficulty. Problem fixed!

I guess the ultimate solution is to either include making sure pip and apio are installed as part of the installation instructions, and/or having the enable/disable functions in Icestudio check for this and provide an error message for that condition. Thanks much for the help!

closed 2016-11-29 by r-ohare

Jesus89, 2016-11-29:
Hi,

Icestudio installs a virtualenv with apio in local $HOME/.icestucio/venv. No internet required because all packages are located inside the package.

I have check the Icestudio commands, and recently a new configuration for drivers enable including:

/usr/local/bin/brew update
/usr/local/bin/brew install libftdi
/usr/local/bin/brew link --overwrite libftdi

Probably the internet connection message comes from brew update, and the app "hang" was the libftdi download. I will investigate that.