FPGAwars/icestudio #112: Drivers en Windows 8.1 64 bits

tonicobi, 2016-12-04:
Estoy trabajando con la placa **ICE40-HX8K** en Windows 8.1 con 64 bits, y no consigo que **Icestudio** detecte la placa. Instalé los drivers **libusbk**, como adjunto en una captura, pero no hay forma.

Sin embargo en Ubuntu funciona todo a la primera.

Muchas gracias.
[image]

Jesus89, 2016-12-04:
Buenas,

Recuerda desconectar y conectar la placa cuando modifiques los drivers. Para una análisis más detallado puedes seguir este tutorial https://github.com/FPGAwars/libftdi-cross-builder/wiki#driver-installation. 

Te recomiendo escribir a la lista: https://groups.google.com/forum/m/#!forum/fpga-wars-explorando-el-lado-libre donde te podrán ayudar también los demás miembros de la comunidad y usuarios de la placa de 8K.

Gracias por la propuesta de las etiquetas, la apuntamos para futuras versiones.

Jesus89, 2016-12-04:
Info:

Continúa en el hilo: https://groups.google.com/forum/#!topic/fpga-wars-explorando-el-lado-libre/PbwgXNE4kcU

closed 2016-12-04 by Jesus89