FPGAwars/icestudio #123: [Windows 10][Bug] Freeze on open

EN10, 2017-01-14:
When opening icestudio-0.3.0-beta-win64 by double clicking on icestudio.exe in Windows 10 x64 (Build 14986), Explorer hangs and icestudio doesn't open. 

I have found a work around:
Go to the icestudio-0.3.0-beta-win64 folder open icestudio.exe through powershell

Shortcut from the icestudio-0.3.0-beta-win64 folder: 
Alt then F then R 
Type: ".\icestudio.exe"

Jesus89, 2017-01-16:
Hi, Icestudio-0.3.0-beta has been validated with W10 x64 Build: 10586. Sometimes in Windows launching a NWJS application the first time takes too long. May be a W10 update has worsened this feature.

Can you open the app by double clicking after opening with the terminal?

We will try to replicate the error.

EN10, 2017-01-16:
I've tested on another machine with build W10 x64 Build: 14393 and seems to work fine.
Although there is a windows warning of unsigned application, but only on the first run.
I'll retest on 14986 later.

EN10, 2017-01-21:
After opening with terminal still not opening by double clicking.
After waiting 5 mins still nothing.

Jesus89, 2017-01-21:
Hi, have you tested the new Icestudio-0.3.0-**beta2**?

EN10, 2017-01-21:
Just tested beta 2 same issue.

On a side note is there a way to change the default board or save the change?
Bit annoying having to change it every time.

Also generally thanks for the good work.

Jesus89, 2017-01-22:
It seems to be a Windows issue. We will try to replicate it.

I agree to save the current board in the app profile. I will create a new issue with the feature.

Jesus89, 2017-02-13:
Hi @EN10,

Can you try this installer https://drive.google.com/file/d/0B_vQY-TVqNFYVHM3TmVjRjJIaDQ/view?usp=sharing in Windows 10 Build: 14986?

Jesus89, 2017-02-14:
Fixed with [0.3.0-beta3](https://github.com/FPGAwars/icestudio/releases/tag/0.3.0-beta3)?

EN10, 2017-02-16:
My windows build updated itself recently it's now 14986.
I downloaded Icestudio-0.3.0-beta3-win64.exe the same issue seems to persist.
Double click the icon, get a loading mouse icon after 5 mins nothing happens except explorer has frozen.
Have to go to task manager to restart explorer.

Jesus89, 2017-02-16:
This is a very weird issue, because the EXE is a windows installer generated with NSIS. Very different from the Icestudio executable.
Can you run any other executable files?

Jesus89, 2017-02-16:
This article may be useful for you: http://windowsreport.com/exe-files-not-opening-windows-10

EN10, 2017-02-17:
I installed 7-Zip Portable for example and the exe works fine:

http://portableapps.com/apps/utilities/7-zip_portable

Jesus89, 2017-03-08:
Can anyone reproduce this bug in Windows 10?

Jesus89, 2017-04-07:
I will close this issue until we can reproduce it for more clues.
Btw, with W10 automatic updates may be the "build 14986" won't be used very frequently.

closed 2017-04-07 by Jesus89