FPGAwars/icestudio #127: Can't execute icestudio properlly after update.

C47D, 2017-01-20:
Hi, thanks for the improvements :)
Just updated icestudio from git and when trying to execute it i got the following screen shoot.
[image]

Just to ask if somebody have the same results.

Thanks in advance.

Jesus89, 2017-01-20:
Hi Carlos,

Please, can you attach your current commit? Also you can launch the developer window: app/package.json -> window.toolbar = true. Launch the app again, and capture the console window.

[image]

C47D, 2017-01-20:
Hi Jesús,
Current commit:
[image]

After setting window.toolbar = true and try to run the application i got the same as the first attachment and here's the console output:
[image]

Thanks for the quick reply, hope the attachments are correct

C47D, 2017-01-20:
Hi,
After changing back toolbar to false i did:
$ npm install
$ npm start

And seems to be working great as always. Here a couple of screen shoots:
[image]

[image]

Thanks again for the continuous improvement :).

I think the issue is solved.

Jesus89, 2017-01-20:
Great. Thanks for testing!

closed 2017-01-20 by Jesus89