FPGAwars/icestudio #137: [Verilog][Code generation] Input not connected to a wire in the Verilog code generation

Obijuan, 2017-02-08:
This bug was first observed by Eladio Delgado. He told me the problem and I started to do some test in order to determinate in which part of the stack was originated.

It seems that in some cases the generated Verilog connections are not working ok

I've isolated the problem in this example:

[image]

It is a D-flip-flop in which the negated output q_n is connected back to the d input to create a T Flip-flop

The generated verilog code is:

```verilog
// Code generated by Icestudio 0.3.0-beta3-dev
// Wed, 08 Feb 2017 06:21:21 GMT

`default_nettype none

module main (
 input vd5501f,
 input vclk,
 output v834948,
 output [0:6] vinit
);
 wire w0;
 wire w1;
 wire w2;
 wire w3;
 assign w0 = vd5501f;
 assign v834948 = w1;
 assign w3 = vclk;
 main_v9ef512 v9ef512 (
  .clk(w0),
  .q(w1),
  .q_n(w2)
 );
 assign vinit = 7'b0000000;
endmodule

module main_v9ef512 (
 input d,
 input clk,
 output q,
 output q_n
);
 reg qi = 0;
 always @(posedge clk)
   qi <= d;
 assign q = qi;
 assign q_n = ~qi;
endmodule
```

We can observe that **the d signal is not connected** when the module ** main_v9ef512**  is instantiated

It should have generated something like this:

```verilog
main_v9ef512 v9ef512 (
  .clk(w0),
  .q(w1),
  .q_n(w2),
  .d(w2)
 );
```

(Or using any other temporal wire between q_n and d)

Obijuan, 2017-02-08:
I am attaching the icestudio project file

[test2-nok.ice.zip](https://github.com/FPGAwars/icestudio/files/760017/test2-nok.ice.zip)

Jesus89, 2017-02-08:
Ok. It seems to be a problem of self-connectivity. Because the q_n wire and the d wire are the same.

Jesus89, 2017-02-08:
Btw, vclk signal mustn't be there. What commit are you using? The current commit generates this output (the bug is still)
 
```verilog
// Code generated by Icestudio 0.3.0-beta3-dev
// Wed, 08 Feb 2017 19:35:02 GMT

`default_nettype none

module main (
 input vd5501f,
 output v834948,
 output [0:6] vinit
);
 wire w0;
 wire w1;
 wire w2;
 assign w0 = vd5501f;
 assign v834948 = w1;
 main_v9ef512 v9ef512 (
  .clk(w0),
  .q(w1),
  .q_n(w2)
 );
 assign vinit = 7'b0000000;
endmodule

module main_v9ef512 (
 input d,
 input clk,
 output q,
 output q_n
);
 reg qi = 0;
 always @(posedge clk)
   qi <= d;
 assign q = qi;
 assign q_n = ~qi;
endmodule

Jesus89, 2017-02-08:
I think it is solved: 59792cd2c955a3e069fa5e9a0e758feb88c265f4

Obijuan, 2017-02-08:
It works! :-)
Thanks!!!

closed 2017-02-08 by Obijuan