FPGAwars/icestudio #154: Icestudio 0.3.0-RC not quitting properly on exit

andrespmt, 2017-04-02:
Desde OS y Windows:

- desde una shell
- lanzo la aplicación (npm start)
- hago algo con ella (supongo que esto no importa)
- salgo de la aplicacion con QUIT

Completed in 144.438s at Sat Apr 01 2017 17:05:34 GMT-0700 (Pacific Daylight Time) - Waiting...

y luego tengo que dar CTRL-C

Jesus89, 2017-04-03:
Buenas Andrés,
Al lanzar la aplicación con 'npm start' se lanza un watcher que está constantemente esperando cambios en los ficheros fuente para relanzar la aplicación. Esto es muy cómodo para desarrollar ya que automatiza esa tarea.
Por ese motivo, cuando cierras la app el watcher continúa funcionando hasta que lo cierras con Ctrl+C es un comportamiento normal.
Un saludo.

closed 2017-04-03 by Jesus89

andrespmt, 2017-04-03:
OK. Gracias