FPGAwars/icestudio #157: Control+U shortcut doesn't work

vrruiz, 2017-04-20:
Test case.
- Open IceStudio and connect the Icezum Alhambra.
- Open example "Four bits fixed binary value".
- Click on the code block.
- Press Control+U.

Expected result.
- Program is compiled and uploaded.

Actual result.
- Program is not compiled nor uploaded.

The shortcut is used by the editor to convert selected text to uppercase.

Tested on Ubuntu with Icestudio version 0.3.0-rc.

C47D, 2017-04-20:
I have the same issue, clicking outside the code editor and then pressing Ctrl + U work as expected.

Jesus89, 2017-04-21:
Hi, this was fixed in dd455c148f62c4b5c13a2728a2a0d1d69b1e1949. Please try the develop branch to ensure that it works properly.

C47D, 2017-04-21:
Hi, It works properly now, thanks for the quick fix.

closed 2017-04-24 by Jesus89