FPGAwars/icestudio #163: Drivers not correctly installed on Ubuntu GNU/Linux 16.04

Obijuan, 2017-06-22:
In Ubuntu 16.04, in some computers, the drivers is not correctly installed

I've checked it with the following commands:

$ ls /etc/udev/rules.d/

and the file 80-icestick.rules should *not* appear (as the driver is not installed)

Now, I execute icestudio and enable de driver. It ask me for the password

If I check again, the 80-icestick.rules file has not been copied into /etc/udev/rules.d

If I install apio and execute it by hand, then driver is correctly installed:

$ apio drivers -e

$ ls /etc/udev/rules.d/
70-snap.core.rules  80-icestick.rules  99-anbox.rules

I should notice that this bug is not present in all the ubuntu 16.04. I've confirmed it in one laptop, but in another one with the same ubuntu version the bug was not present

Jesus89, 2017-06-23:
It happens only in the AppImages, because can not copy 80-icestick.rules file:
```
cp: cannot stat '/tmp/.mount_DvE0fZ/usr/bin/resources/config/80-icestick.rules': Permission denied
```
Solved using echo instead of cp.

closed 2017-06-23 by Jesus89

Jesus89, 2017-08-13:
c7898d4fff7aa5e518cc658f48487390511f065a

commit ab0d0593 (2017-06-23): Fix Linux drivers enable. Closes #163