FPGAwars/icestudio #168: [enhancement] allow SRAM only programming

chriscamacho, 2017-07-25:
allow uploading only to fpga sram and not to flash, I regularly use iceprog -S as it is *much* faster especially while testing, obviously once tested you'd want to revert to programming flash...

Jesus89, 2017-07-27:
Icestudio uses [apio](https://github.com/FPGAwars/apio) as backend. Any enhancement in the execution of the toolchain must be implemented first in apio.

Then it can be implemented in Icestudio. Since Icestudio supports multiple boards with different programmers, do you have any preference in how to implement the UX/UI of this feature?

chriscamacho, 2017-07-27:
in my makefile for icestorm I have two targets prog and prog-ram

in a similar manner in the menu you could have "upload" and "upload to sram" it should be noted on my lattice breakout board, you have to remove one jumper and rotate a block of two other jumpers before you can upload to sram and this then prevents upload to flash...

for usability reasons a better way might be to have a project setting say a check box "upload to sram instead of flash" (for the boards that can do it) and then only have an upload menu item by default upload should go to flash 

The circuit diagrams for the lattice ice40hx8k are readily available and it might be worth looking at how these jumpers work as some other boards might be able to support this functionality with a minor "tweak" - for all I know some others might be able to do it already

Jesus89, 2017-07-27:
Great. I like the "Upload to SRAM" checkbox. We want to see if it is a project setting, a user setting and where to put this setting: Tools, Edit, ...

Thanks for the suggestion. It will be implemented in future releases.

chriscamacho, 2017-07-28:
I'd have thought a project setting, but also consider a pop up when selected to warn the user to change the jumpers, I know after leaving my board for a while I couldn't program flash forgetting about the jumper settings but don't know if there is anyway to infer how the physical jumpers are set ?

nmz787, 2020-07-12:
I just started trying icestudio tonight (linux nightly).... after cutting the traces on the icebreaker and successfully getting SRAM upload working on the command-line. Is this feature really still unsupported, or has this issue simply not been updated?

nmz787, 2020-07-12:
this `apio` PDF says their upload command supports `-s` https://readthedocs.org/projects/apiodoc/downloads/pdf/latest/

```
-s, --sram
Perform SRAM programming. Only available for iceprog compatible boards.
```

how can I quickly hack this to work? Clicking upload has now frozen Icestudio and there's no **cancel** button

nmz787, 2020-07-12:
ok, after a little digging, I found the `/tmp/icestudio` prefix is where active projects are built. Since I only had one project open, after `Build`ing I was able to just `cd` into the latest directory... i.e. `cd `ls -tr /tmp | grep icestudio | tail -1``

then `iceprog -S hardware.bin` to shove it into the icebreaker's SRAM

Georgschmied, 2020-09-20:
Hello, i tried to program into SRAM but I don't know how to change the apio settings, can somebody help me please I'm using Win10