FPGAwars/icestudio #169: Add testbench editor + Simulation

Jesus89, 2017-08-05:
Edit timescale and signal sequences.

tilk, 2018-06-20:
I'm seriously interested in having simulation capability in icestudio. I'm planning an undergrad course on basic digital logic for CS students, and I'm looking for a conceptually simple toolchain for graphical and Verilog logic circuit design.

lmcapacho, 2024-12-03:
I'm very interested in working on this feature. I was wondering if there is currently any work in progress related to this feature request. I also think that issue #716 might be connected to this.

cavearr, 2024-12-04:
Hi @lmcapacho ! We are improving Celery these days and it will improve support for gtkwave more deeply than now. In the next few days we will release icestudio wips with these new features.

In parallel, I am working on new features related to simulation and testbenches that could be released at the end of January (this month I plan to release a very important update with new boards until the end of the year and then focus on the testbenches). Do you want to work on it? If you want to work with us, tell us your ideas and we could organize the work around them if they make sense.

Thank you!

lmcapacho, 2024-12-04:
Hi @cavearr, 

Yes, I am interested in working on this feature. I was thinking of including an editor for the testbenches that allows modifying the testbench generated by Icestudio. Another idea is to add a "Simulate" option in the Icestudio menu, which would call the `apio sim` command and run GTKWave with the result.

cavearr, 2024-12-04:
Hi @lmcapacho, the icon with the Apio sim command callback will be ready in a few days, I'm working on it now. On the other hand, the editor, you could create a plugin for this.

I've been working on a new editor covering Verilog and the testbench editor (and a GUI for node-based testbeds) for a while, but this will be available in two or three months (I'll try to post my roadmap soon) .

But if you would like to try to make an alternative editor or a first version of an editor, tell me more and I might be able to help you on the path to doing so.

This month I plan to launch a plugin that works as a "play store", the idea is that users can download blocks, plugins.... like Visual Studio and its ecosystem, I think this is the best place for your editor. In the meantime, you can get started, the plugin development is really clean and independent. If you go this way, an expansion of the functions you may need might be necessary, but I could provide it.

Like I said if you want to do it, go for it! and I try to help you.

If you would prefer to collaborate with tasks that I already have in progress, tell me and I will try to organize small tasks that you could start with and we will see.