FPGAwars/icestudio #171: Make it compatible with TinyFPGAs

Syn0rLife, 2017-09-11:
Hi,

I think the following project is the most hopeful open source FPGA implementation at this point : 

https://github.com/tinyfpga/TinyFPGA-B-Series

I bought one (B2, based on iC40LP8K) and would be very pleased to help in any way I can to integrate it in the iCEEStudio project (making some tests for exemple, or completing files if i'm guided).

Thank you

Jesus89, 2017-09-11:
Hello,

First step is to add this board in [Apio](https://github.com/FPGAwars/apio). I will think how to integrate its python programmer. It would be nice to add board checking for this TinyFPGA, is it a FTDI based board?

Second step is to integrate this board in Icestudio: [pinout](https://github.com/FPGAwars/icestudio/tree/develop/app/resources/boards) and drivers/configuration for the board.

If you want to contribute, you can try Apio integration, and also the pinout (pcf) for Icestudio.

Regards!

tinyfpga, 2017-09-30:
Hi Jesus, I am the TinyFPGA developer.  The board is not FTDI based, it implements it’s own USB device in the FPGA fabric as a bootloader.  When ai user design is running the USB bootloader is no longer in the fabric and is stored in the SPI configuration flash.

If anyone is working on this already I would love to help.  If not, I’ll take a look at how much work it would be to integrate myself and submit a push request.

Using the pyserial library you can identify the board by its USB VID and PID combination.  The board has its own Python library for programming it.

Syn0rLife, 2017-10-01:
Hi,
I didn't make anything yet, and it seems too much complicated for me so I'll let you start.
I'ld be a very intensive user of these tools if you achieve it ^^

tinyfpga, 2017-10-08:
@Jesus89, I have updated a local repo of apio as well as icestudio with the TinyFPGA B2 board information.  How do I setup icestudio to use a custom build of apio and not one that it can automatically install?

Jesus89, 2017-10-08:
Hi @tinyfpga,

Icestudio uses a python virtualenv to install its own apio version. You can create the virtualenv and install all the packages in the menu Tools > Toolchain > Install. Then you should have an `.icestudio` directory in your home path with a `venv` directory inside.

You can activate the virtualenv by executing this console command in the `.icestudio` directory: `source venv/bin/activate`. Now you can install apio from the sources inside the virtualenv `pip install -U .`

I use Linux to develop both projects, but it should be working also for windows. Please, paste the error output in a pastebin or something like that to check what is going on on windows.

Jesus89, 2017-10-08:
+ you can use also an external apio (installed in your machine) by adding its path [here](https://github.com/FPGAwars/icestudio/blob/develop/app/package.json#L23). For example `"external": "/home/jesus/.local/bin/apio"`. In this way you don't need to activate the Icestudio's venv.

tinyfpga, 2017-10-08:
@Jesus89, I pointed it to the apio that I built and that worked for running "Tools -> Verify" in the Icestudio GUI.  However, when it ran "Tools -> Build" it complained that it didn't recognize the board.  I worked around this by copying my apio.exe binary to the virtual environment and replacing the one that was there.  Once I did that both Verify and Build worked.

However, when I try to run "Tools -> Upload" I receive an error in the GUI and I'm not sure how to get more detailed information: "Error: expected a string or other character buffer object"

[image]

Jesus89, 2017-10-09:
Hi. When you execute Toolchain > Upload this runs the commmand "apio upload -b yourboard" so the error is launched by apio.

I recommend you to get the board working with apio and then Icestudio. You can push your apio changes in a feature-branch in your fork so I can take a look.

Jesus89, 2018-01-11:
Closed in https://github.com/FPGAwars/icestudio/pull/193.

closed 2018-01-11 by Jesus89