FPGAwars/icestudio #172: Option to export Verilog using usable port names

set-soft, 2017-09-23:
When exporting Verilog add an option to name top-level signals using the name of the input/output block instead of some bizarre magic name.
Why? To be able to simulate the block using an external testbench.
Currently we must figure oyut what's each of the magics.

Jesus89, 2017-10-07:
Hi Salvador,

Those bizarre magic names are internal uuid codes required to ensure that the generated verilog code is always correct. Because you can have different I/O with the same name (label).

Until the simulation is integrated in icestudio you can export an automatic generated testbench file (and also a gtkwave file) to enable manual simulation. In this file you have the map between the uuid codes and the real names. Is this useful for your purpose?

Regards.

Jesus89, 2018-05-22:
Closing this. Open if you need more information.

closed 2018-05-22 by Jesus89