FPGAwars/icestudio #175: [Bug] Closing App with multiple windows

ChuxMan, 2017-10-22:
When you open a window inside a first opened (file --> new) and you close the first one, the entire program close it up without saving the second.

Detected in mac. Not tested on the other platforms.

[image]

Jesus89, 2017-10-22:
Hi. I have tested MacOS 10.7 and GNU/Linux Ubuntu 16.04 but I can not reproduce the bug. What  version of MacOS are you using?

ChuxMan, 2017-10-22:
This is High Sierra

[image]

And IceStudio is the previous version, I'm going to download the lastest one and check it again.

[image]

minicatsCB, 2017-12-14:
I have experienced this too, but not always, only in some cases, in Ubuntu (in VirtualBox).

charliva, 2019-02-01:
Hi! I'm working on the problem "Low resolution in mac retina when opening a new window". # 267, is related to this. When I make a pull request to 267 this will be solved quite likely.

Jesus89, 2019-02-02:
We need to verify if this is fixed with the latest develop (https://github.com/FPGAwars/icestudio/pull/303)

PhilippvK, 2019-02-02:
Thank you for your work.

Unfortunately the App crashes if I perform the following steps on OSX:
- start ice studio
- open a new window from the menu (opens as new tab)
- close the old window/tab
- open a new window from the menu
- CRASH

But I am happy that #267 is fixed now!

charliva, 2019-02-04:
I'm working on it, in the next few days I think the error will be corrected.

charliva, 2019-02-04:
Hi! i fix this bug and i'm doing a pull request:

https://github.com/FPGAwars/icestudio/pull/304

When you try it, tell us if works fine ( I've tried it in a mojave macosx and it seems that everything is fine)

PhilippvK, 2019-03-11:
Seems to be fixed on OSX now. Thank you!