FPGAwars/icestudio #176: [Bug] Lost focus on a big project

ChuxMan, 2017-10-22:
When you are in a big project 

[image]

and zoom in on a bottom corner, 

[image]

and then you zoom out on the top

[image]

you lost the focus of the top of the project, and is no longer accessible

Again, tested in mac, not other platforms.

ChuxMan, 2017-10-22:
I miss a photoshop navigator-like to move on the project or something similar. Why not scrollbars?

Jesus89, 2017-10-22:
Hi @ChuxMan:

The zoom behavior is the same than Inkscape. The zoom point is the cursor. The issue here is that the scrollbars shouldn't be here. Btw, what version of MacOS are you using?

You can move the background with Ctrl + L mousedown or R mousedown. And also select Edit > Fit content (Ctrl+1) to center and scale the design (this also happens when the window is resized). You can check the shortcuts here: https://github.com/FPGAwars/icestudio/wiki#mouse.

By now everyone is uploading its own blocks in personal repositories. You can create an Icestudio Collection (https://github.com/FPGAwars/icm) and share your repo in the group.

cavearr, 2020-01-29:
Fixed on v5.1.0 dev branch

closed 2020-01-29 by cavearr