FPGAwars/icestudio #191: [Feature] Blocks replacement

Obijuan, 2018-01-05:
Currently, if you want to change one block for another (for example an updated block), you have to delete that block and insert the new one. The problem is that you lost all the connections. This is a little bit painful when your design is complex.

In addition, the replacement feature makes the exploration process of newbies much easier. Here are some examples, starting from the simplest one

* Changing the state of a led:

This is the circuit for turning a led on:

[image]

With the replacement feature, you pick a new block, for example a bit 0:

[image]

put it over the block you want to change

[image]

click the mouse button and the block is replaced

[image]

If you replace a block by error, you can use ctr-z (undo) to move back to the previous state

Now imagine you want to  make the led blinks. You just replace the bit by a heart. Pick the heart and put it over the block and click the button

[image]

[image]

I really think that this feature will make a difference in the educational proyect. I can image an icestudio sheet, with a simple circuit (like the hello world before) and different blocks on the left that change the behaviour on the led. The student just drag the blocks to test what they do with the led

[image]

There are many details that should be cover. Let's use this issue to document them

Obijuan, 2018-01-05:
It would be great if the replacement works also with code blocks:

[image]

[image]

It brings the question... if the blocks have different names for their outputs, how should the replacement proceed? This is something that we should solve by using it. We will discover the rules as we test it

One rule might be:  
1. If the two block have the same number of outputs, regardless of their names, the replacement operation will be done.   

This rule will cover the above case

Obijuan, 2018-01-05:
One important thing is that the wiring should remain the same after the replacement. I know it should be hard to implement. Let's say we have this circuit

[image]

And we want to replace the 1 by the heart. After the replacement has been done, the wiring should stay the same:

[image]

Jesus89, 2018-01-20:
Hi!

I have created a PR for this: https://github.com/FPGAwars/icestudio/pull/198.

Now you can replace Generic, Code and Info blocks. When the block is replaced, the wires are replaced for each side (left, right and top). First: it tries to match the same name & size connections. Second: if there are ports with the same name, they are replaced by position.

The algorithms has also two steps: create a portsMap between the lowerBlock and the upperBlock, and try to replace the connected wires using this portsMap.

It also preserves the wire's vertices and Undo/Redo works for move&replace and create&replace blocks.

The only thing missing is the match between ports with different name. I need to figure out how to define this behavior to be consistent with the current algorithm.

Jesus89, 2018-01-21:
I have added the following rule. This rule is applied after the first names&size match:
> If there is no match replace the connections if the port's size matches ignoring the port's name.

This is an example:

[image]

Obijuan, 2018-01-21:
That's awesome! Thanks!  :-)

Obijuan, 2018-01-21:
I think that this issue can be closed.  We all can test it and create a new issue when a bug is found

closed 2018-01-21 by Obijuan