FPGAwars/icestudio #196: [Bug] Icestudio close when you press CTRL+U

ximocat, 2018-01-16:
Cuando cargas y activas una de las últimas colecciones del curso de Obijuan, por ejemplo, Academia-Jedi-HW-10 (o la del tutorial 11), copias y pegas un componente y pulsas CTRL+U se cierra icestudio. No lo he probado con todos los componentes, pero con los que seguro que sucede es con la tortuga (retardo) y con los controladores de servos.

Para reproducirlo:
1- Abrir icestudio
2- Cargar y Activar la colección: https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/raw/master/wiki/Tutorial-10/Collection/Academia-Jedi-HW-10.zip
3- Poner en el panel una tortuga (retardo) o un controlador de servo
4- Copiar y pegar el componente
5- Pulsar CTRL+U

Se cierra Icestudio

Probado en dos ordenadores diferentes en windows 7 y windows 10. Icestudio versión 0.3.1

Jesus89, 2018-01-16:
Buenas @ximocat.

Has probado otras secuencias que den el mismo error? para tener más pistas. Tienes la placa conectada y los drivers instalados? Puedes guardar el ICE antes del paso 5 y adjuntarlo?

En Linux con la misma versión no consigo reproducirlo. @Obijuan puedes probar en Windows VM? Gracias.

ximocat, 2018-01-16:
Me di cuenta de este error al hacer los tutoriales de Obijuan. Sucede tanto con la tortuga como con los servos. Tanto con la placa conectada como sin conectar y los drivers sí están instalados.

[image]

Pongo la tortuga
[image]

Copio y pego
[image]

Y al pulsar CTRL+U se cierra

El archivo ICE (comprimido):
[Untitled.zip](https://github.com/FPGAwars/icestudio/files/1635102/Untitled.zip)

ximocat, 2018-01-16:
He grabado un video. En este caso no tengo la placa, pero con la placa conectada pasa lo mismo.

Da igual si los componentes están conectados o no, sucede igual

https://youtu.be/6-aDcH89HLg

ximocat, 2018-01-16:
Creo que sucede en la colección del tutorial 10 y la del 11, así que supongo que algo tendrá que ver la colección

Jesus89, 2018-01-16:
Gracias por la información @ximocat!

Obijuan, 2018-01-16:
Consigo reproducir el bug en la máquina virtual con un Windowws 7
Voy a  hacer más pruebas y comento

Obijuan, 2018-01-16:
La ventana se cierra al darle a cualquiera de las opciones del menú Herramientas: Veriificar, sintetizar y cargar (no es debido a los atajos)

Obijuan, 2018-01-16:
Confirmo que con la versión de development también ocurre.

Lanzo icestudio con npm start:

[image]

Coloco la tortuga (de la colección 10) y hago copy&paste

[image]

Ejecuto cualquier opción del menú Herramientas y se cierra la ventana. En el terminal se puede ver lo que aparece: no hay un error, sino un Done

[image]

Jesus89, 2018-01-16:
Gracias Juan!

El output de la consola es normal. No aparece ninguna información extra ahí.

Ejecuta Icestudio con las opciones de consola activadas. Para ello pon este flag a true (https://github.com/FPGAwars/icestudio/blob/develop/app/package.json#L15) y ejecuta npm start. En Icestudio ahora aparece la barra de herramientas. Pulsa en la "hamburguesa" y se te abrirán las Developer Tools. Ahí en "Console" debería aparecer información.

He creado la rama `196-windows-bug` para ir trabajando en la solución.

Un saludo.

Jesus89, 2018-01-16:
Después de unas pruebas hemos visto que puede ser por el tamaño del fichero. Como no podemos visualizar las trazas vamos un poco a ciegas.

La siguiente prueba que se me ocurre, para ver si apio está implicado o no, es realizar la misma operación pero al final en vez de "Tools > Upload" ejecutar "File > Export > Verilog" y ver si sucede lo mismo.

Seguimos a la caza del bug...

Jesus89, 2018-01-21:
Buenas! Hoy he probado Windows en una VM y he encontrado el bug.

Se trata de un **bug en node.js**, concretamente en la versión que acompaña a NW.js 0.12.3.

El error consiste en una inconsistencia del módulo `Buffer`. El problema tiene dos partes:
- Al hacer **paste** de un bloque "grande" (300kb) se ejecuta una instancia de Buffer que genera overflow de la variable estática `poolOffset` ya que supera el tamaño del pool. Con bloques pequeños el bug no sucede ya que no hay overflow. Con bloques grandes la variable `poolOffset` se vuelve negativa. ([code](https://github.com/xavi-/node-copy-paste/blob/9897be123ca03e4946325270dcc80a5736a36cc7/platform/win32.js#L20))
- Al **generar el PCF** se produce el error que cierra la aplicación como consecuencia del primer paso. El PCF resultante del diseño se escribe en un fichero temporal `main.pcf` utilizando la función de nodejs`fs.writeFileSync`. Esta función, antes de escribir el fichero, convierte el contenido que se pasa por parámetro de string a Buffer y es ahí donde está el error, porque el contenido del fichero mas el offset da un valor negativo de posicionamiento. Esto no sucede con el fichero `main.v` porque el contenido de este fichero mas el offset da un valor positivo.

Aparentemente estas acciones independientes tienen relación a través del objeto `Buffer` que es el causante de este curioso bug.

He visto que ha cambiado bastante la implementación de `buffer.js` en nodejs. Se ha mejorado la consistencia del módulo, así que una posible solución es actualizar la versión de `NW.js` a la más reciente y comprobar que funciona correctamente en Windows.

Un saludo.

cavearr, 2020-01-29:
Fixed on v0.5.1 dev branch

closed 2020-01-29 by cavearr