FPGAwars/icestudio #197: Icestudio window contents cannot exceed 1920x1080?

janrinze, 2018-01-18:
When full-sizing the icestudio window the contents get clipped to 1920x1080.
On a 2560x1440 monitor this results in a white area to the right and bottom of what is displayed.
The window content seems to be clipped to 1920x1080.
I have search for these values in the code but have not found specific references to such clipping.
Are there settings somewhere that restrict the window contents to 1920x1080?

Jesus89, 2018-01-18:
Hi!

Icestudio is using NW.js 0.12. Try adding `max_width` and `max_height` [here](https://github.com/FPGAwars/icestudio/blob/develop/app/package.json#L10-L19).

Btw, I have a ticket to update NW.js to the latest version. This update may solve the issue.

janrinze, 2018-01-18:
Hi :-)

Adding the max_width and max_height seems to make no difference.
Hopefully the update for NW,js will help.

charliva, 2019-01-30:
I'm doing a pull request that fix it, while they are approbing it you can apply the patch

closed 2019-02-02 by Jesus89