FPGAwars/icestudio #202: Unable to upload to Blackice I

andreacampanella, 2018-02-06:
I have an usb cable connected to the nearest USB to the arm, the jumper is closed , icestudio can't find the blackice, any idea?

Jesus89, 2018-02-17:
Hi @andreacampanella can you add more information. What version of Icestudio and OS are you using?

Thanks.

Jesus89, 2018-02-17:
Hi, I found you in this thread: https://forum.mystorm.uk/t/testing-blackice-with-icestudio/345.

So, Ubuntu 16.04 and Icestudio 0.3.2-beta. If the message is `Board Blackice is not connected` means that the board is not detected at the physical level.

Can you install apio: http://apiodoc.readthedocs.io/en/latest/source/installation.html and execute the following commands?
- `apio install system`
- `apio system --lsusb`
- `apio system --lsserial`
- `apio system --lsftdi`

Thanks.

andreacampanella, 2018-02-19:
ok , for some reason I don't understand the version installed via the debian repo is : 
apio, version 0.2.4.1
I will try to compile the latest one...

Jesus89, 2018-02-19:
Yes, the debian package is not updated. Please install it with `pip`.

The latest version of APIO is  0.3.0.

andreacampanella, 2018-02-19:
andrea@Andrea-WS:~% apio system --lsusb                         
062a:5918 (bus 2, device 4) path: 1.6
0483:df11 (bus 2, device 6) path: 1.5
1b1c:1b07 (bus 2, device 3) path: 1.4
1d6b:0003 (bus 4, device 1)
1d6b:0002 (bus 3, device 1)
8087:8000 (bus 2, device 2) path: 1
1d6b:0002 (bus 2, device 1)
8087:8008 (bus 1, device 2) path: 1
1d6b:0002 (bus 1, device 1)

andrea@Andrea-WS:~% apio system --lsserial
Number of Serial devices found: 0

andrea@Andrea-WS:~% apio system --lsftdi
Number of FTDI devices found: 0

andrea@Andrea-WS:~% apio --version
apio, version 0.3.0

andreacampanella, 2018-02-19:
I found it , the board was in DFU mode because the jumper was in the wrong place, is working now, thnak you for the patience :)

Jesus89, 2018-02-19:
Perfect!

lsusb and lsserial  commands show that the board is not connected (0483:5740). It makes sense.

Btw, could you add a picture of the jumper in the correct position in order to have more information for future issues? Thanks.

Closing this ticket.

closed 2018-02-19 by Jesus89

andreacampanella, 2018-02-19:
This is a picture of the board, with the right USB port connected and the jumper in the right position.

[image]