FPGAwars/icestudio #205: Add updated `setuptools` and `wheel` packages

Jesus89, 2018-02-17:
These packages must be updated to install SCons. So:

- [ ] Add offline packages to the app bundle
- [ ] Install these packages before `apio` in the icestudio virtualenv

ColRod, 2018-02-19:
Tras ejecutar en consola:
pip install -U setuptools
pip install -U wheel
y despues de que la maquina actualizase, sigo con el mismo problema que me impide instalar Toolchain en 0.3.2-beta
[image]

Jesus89, 2018-03-03:
Hola @ColRod.

Para actualizar esos paquetes en icestudio tienes que activar el virtualenv (`~/.icestudio/venv`) e instalar ahí los paquetes. En Linux sería:

```
source ~/.icestudio/venv/bin/activate
pip install -U setuptools
pip install -U wheel
```

De todas formas la solución que voy a implementar es tener los paquetes actualizados descargados en local e instalarlos antes que apio.

Jesus89, 2018-04-12:
[Apio v0.4.0b1](https://github.com/FPGAwars/apio/releases/tag/v0.4.0b1) cambia el sistema de instalación de scons por lo que ya no requiere que estos paquetes estén instalados.

Sin embargo, dejo las funciones de pre-instalación de paquetes por si en el futuro fuera necesario instalar algún paquete externo de Python.

closed 2018-04-12 by Jesus89