FPGAwars/icestudio #220: unable to install the toolchain for TinyFPGA B2 on macOS 10.13.4 and icestudio 0.3.2

dennyalbert, 2018-04-22:
Install process stop working with „Create virtualenv…“.

[image]

Jesus89, 2018-04-22:
Hi, please try [0.3.2-beta](https://github.com/FPGAwars/icestudio/releases/tag/0.3.2-beta) version.

As soon as I get a Mac I'll fix the issue.

Thanks!

blkhawk, 2018-05-12:
I have the same problem - There are a bunch of "Operation not Permitted" errors on some .so files in ~/.icestudio/..  The files exist and seem to have the right permissions.

Jesus89, 2018-05-18:
Maybe related to this commit: https://github.com/FPGAwars/icestudio/commit/c0e66a2eec47a1ac3e94346087670a0522d9b14f

We have tested that 0.3.0 version works. The one using `tar.gz` for the virtualenv instead of `zip`.

Jesus89, 2018-05-19:
Hi @blkhawk.

I finally get a High Sierra Mac and fix the issue. I have replaced the installers here: https://github.com/FPGAwars/icestudio/releases/tag/0.3.2.

Please, check that this is also fixed in your computer.

Regards.

closed 2018-05-20 by Jesus89