FPGAwars/icestudio #228: A feature of markdown doesn't work

maxezek, 2018-05-13:
The integration of HTML and markdown seems perfect but I find a type of text in markdown that it is not translated by Icestudio

* [ ] This is interesting for describing a task.
* [x] This represented a finished task.

But in Icestudio * [ ] it is not converted in
* [ ] .

Jesus89, 2018-05-13:
Hi @maxezek 

Icestudio uses [marked.js](https://github.com/markedjs/marked) to generate the HTML view from the markdown code so in order to add this new feature it should be supported first in `marked.js`.

Jesus89, 2018-05-13:
I have just found [this library](https://www.npmjs.com/package/roaster) that uses marked for Markdown convertion and [Task List JS](https://github.com/gjtorikian/task-lists-js) for the checkbox list.

So we could use the same library to implement this feature in Icestudio :blush:

Jesus89, 2018-05-17:
Hi. I have implemented this feature and other improvements for the Info block (#235).

closed 2018-05-17 by Jesus89