FPGAwars/icestudio #229: Improve include files

Jesus89, 2018-05-14:
- [ ] Extract files in the compiler (optimization)
- [ ] Implicit include for `.vh` files
- [ ] Allow double quotes `" "` in explicit includes
- [ ] Allow relative file paths `"../libs/math.v"`

**Implicit**
- `"rom.list"` (not within a comment)
- <code>`include "lib.vh"</code> (not within a comment)
- No implicit include for `.v` files

**Explicit**
- `// @include "rom.list"` | `// @include rom.list`
- `// @include "lib.vh"` | `// @include lib.vh`
- `// @include "../libs/math.v"` | `// @include ../libs/math.v`

rjsm1955, 2018-08-18:
Firstly, thanks for the fantastic work on IceStudio - I'm new to this and it sure seems it will make working with FPGA's a pleasurable experience.

I have some pre-existing verilog .v files I'd like to experiment with in IceStudio.  I've tried including them using // @include myfile.v inside a code block and verification proceeds ok - but the build does not appear to result in any FPGA resources being allocated and although the upload step appears to work there is no activity on one of the on-board LEDs as there should be (FPGA board is an ICE40-HX8K).  If I export the generated verilog from IceStudio and examine it I just see the // @include line within the verilog module definition but nothing else.

However if I now enter exactly the same code manually into the code block the verify, build and upload steps all work fine with the expected LED activity.

What could be the problem here ?  I imagine this should be straightforward and there must be an obvious solution.  I've also tried copying my verilog and then pasting into the code block but the IceStudio code block editor doesn't seem to retrieve pasted text from the clipboard.  Is that something that can be supported in the future ?

marcio-ao, 2018-12-27:
@rjsm1955: Did you instantiate the module? For example, here is a test I am working on where I include "lcd.v" and then instantiate it:

[image]