FPGAwars/icestudio #231: Soportar acentos, eñes y espacios en los nombres de bloques

javierpuche, 2018-05-14:
¿O es una limitación de verilog?
Gracias en cualquier caso :-)

Jesus89, 2018-05-17:
Buenas @javierpuche.

La limitación de los nombres está realmente en los puertos y parámetros de los *bloques de código* porque se utilizan directamente en el código verilog.

En verilog los identificadores deben comenzar por `A-Za-z_` seguidos de hasta 1024 `A-Za-z_$0-9`.

Sin embargo los nombres de los bloques de Entrada/Salida/Constante podrian llevar cualquier caracter (exceptuando [] para no confundir con los buses) ya que en verilog se utiliza el identificador del bloque en vez del nombre asignado.

He actualizado las expresiones en este commit: 1deded1e2b50d834169dd5ab52e1bd3c1863affc.

La limitación por los espacios viene de simplicar la declaración de múltiples puertos a la vez, por ejemplo `a ,  b` -> `a` y `b`. Sin embargo he visto una forma mejor de implementar esto que permita espacios en el nombre, por ejemplo `a b , c` -> `a b` y `c`. Lo he implementado en este commit: d588b120e84bb7dfd48e535581f3ec5d9a2a1c6f.

**Conclusión**: después de darle una vuelta los puertos de los bloques Entrada/Salida/Constante/Memoria pueden tener nombres con cualquier caracter. Los puertos y parámetros de los bloques código deben cumplir con el estándar de identificadores de Verilog.

Un saludo.

closed 2018-05-17 by Jesus89

javierpuche, 2018-05-22:
> Buenas @javierpuche[1].
>   He actualizado las expresiones en este commit: 1deded1[2].
> implementado en este commit: d588b12[3].

¡Mil gracias! :-)
Saludos.

Vínculos:
---------
[1] https://github.com/javierpuche
[2]  
https://github.com/FPGAwars/icestudio/commit/1deded1e2b50d834169dd5ab52e1bd3c1863affc
[3]  
https://github.com/FPGAwars/icestudio/commit/d588b120e84bb7dfd48e535581f3ec5d9a2a1c6f
[4] https://github.com/FPGAwars/icestudio/issues/231#issuecomment-390049625
[5]  
https://github.com/