FPGAwars/icestudio #234: Cannot Upload on Windows 10

EN10, 2018-05-16:
I get this error when I try uploading

[image]

Interface 0 is using libusbK

[image]

RGD2, 2018-05-25:
Happened for me too -- but only when I had a second FTDI device plugged in at the same time.
Unplugging the second FTDI thing fixed it.

EN10, 2018-05-25:
the only other thing I have plugged in is a keyboard and mouse! :(

TimRudy, 2018-07-06:
I have the same error - Windows 10, USB 2.0

TimRudy, 2018-11-11:
This problem indicates that Windows 10 cannot quite handle:
  1) your USB hardware, and
  2) the drivers it uses (libusbK is just a generic, long-standing Windows driver), and
  3) the FTDI driver for IceStick or IceZUM Alhambra or other FTDI device

There is a work-around:

Get yourself a USB hub and try again - USB 3.0 is known to work, not sure about 2.0. Very helpful guy David S. told me this. The hub I tried was bus-powered (simple, inexpensive type). You do not need a separately-powered hub. I literally plugged my son's USB 3.0 hub into my PC's USB 2.0 - take note - so even the old computer ports you already have should work. USB version 2.0 or 3.0 does not matter in that sense. My computer is kinda old and has been through different things in its lifetime. I also note that my computer has lots of 3.0 ports, and 2.0, and 1.0. I tried THEM ALL. Somehow the new USB port in the USB hierarchy - not anything to do with power or voltage, I guess - overcame what is actually a hardware and wonky Windows issue.

So I have my IceZUM Alhambra plugged in the front USB port on my new monitor, and the Zadig driver installation, and connecting to it worked, first time, as it should!

Here's the not-so-easy (perhaps) work-around:

You can use your FPGA through a virtual Windows 7 PC, as other people did who have had this problem. (Same hardware. Same USB that wasn't working, who would have thought.) I'm going to recommend do not try a Windows 8 VM but go straight to Windows 7 because 8 may waste your time, and wasted my time.

Is Issue #234 described as a problem or a bug? Please comment if you experience this, because it is a concern if Icestudio adoption is slowed down and is frustrating for Windows users. 

It may be hard to debug the driver error and try to fix this in terms of item (2) above, libusbK interfacing requirements. (A bug to open with Microsoft?) @Jesus89, is there any other news on this?

Background and research:

I tried every possible driver combination and order of switching them around to Interface 0 or 1 (thinking that history of what I do and "state" might make a difference but that is exceedingly unlikely), so I caution people who want to debug/overcome this: the connection should either work with one or two attempts with Zadig, applying a driver (e.g. libusbK, WinUSB) to Interface 0 and 1, or it ain't going to work!

Use "find_all.exe" to quickly test and get:
  Number of FTDI devices found: 2
  Checking device: 0
  ftdi_usb_get_strings failed: -4 (libusb_open() failed)

Use Windows Device Manager "Scan for hardware changes" instead of repeatedly unplugging and replugging your device.

Therefore not recommended:
- All driver install and uninstall combinations that can be thought of
- All different ports on your PC, versions USB 2.0 or 3.0
- Manual downloaded FTDI drivers, combinations of USB Serial Converter A, USB Serial Converter B, observing your Device Manager and thinking of ways to trick Windows
- 64-bit, 32-bit Icestudio or other things, versions of drivers
- Downloading a newer version of Zadig (trust me, that was exceedingly unlikely)
- Downloading an experimental driver which was actually dangerous (I read that someone had wrecked their running OS which I did not do just because of the version I had downloaded!)

Some technical links:

Developer point of view, for FPGAWars - Some expertise may be available through people on this project:
https://github.com/pbatard/libwdi/wiki

User understanding point of view - warning that nothing here worked, so that's why you are at Issue #234:
https://github.com/libusb/libusb/issues/370#issuecomment-365919954
https://github.com/cliffordwolf/icestorm/issues/141
https://learn.adafruit.com/adafruit-ft232h-breakout/windows-setup

The issue is to be blamed on Windows 10 itself, seems like, and maybe back to Windows 8 - The USB 3.0 driver stack and its internal design was changed starting at Windows 8, and there was also a USB design change at Windows 10: https://blogs.msdn.microsoft.com/usbcoreblog/2015/07/29/what-is-new-with-serial-in-windows-10.

Cheers!

RGD2, 2018-12-18:
Hit this again on a different win10 machine. (release 0.3.3 win64)

Steps to reproduce:

1. Have *ANY OTHER* FTDI usb-rs232 serial adaptor plugged in, on or off hub.
2. Error message appears as above during upload to eg icestick
3. Unplug that other FTDI usb-serial thing
4. icestudio upload works again
5. plug usb-serial adaptor back in
6. broken again!

Seriously. This seems like it's just because icestudio expects to get just one ftdi *device* on a computer at once, and there being more than one maybe whatever ftdi library used returns a list of objects rather than just one object, so it's only not-broken when there's just one ftdi usb-serial on the computer.

This is pretty bad. It's fairly common to need to talk to rs232 to some other thing, whilst wanting to be able to run icestudio in a lab environment.

It might be a separate issue to the above, but it seems like it could be the same one, since the errors' the same. Perhaps sometimes windows or the ftdi lib or whatever gets 'stuck' with returning a list of one item, rather than just the item like it otherwise defaults to? Just a guess.

Windows does try to keep track of usb devices on a per-port basis, so just plugging it into somewhere else when it starts doing this bug *without* other ftdi usb devices seems to fix it. (ditto for using a hub, since in both cases it counts as an 'entirely new device' so far as Window's usb driver system is concerned. And yes this means you can have same device change drivers by shifting back and forth between ports, if you want). 

AFAIK no other OS actually does this, and I don't think it's actually part of the USB spec either (probably MSFT doing their own thing, like usual. They helped write the USB spec, must have just 'forgotten' to include that particular extension.... or maybe they did and just made it 'optional', like needing to 'find/install drivers' is).

RGD2, 2018-12-18:
Ie, must get either a list of lists of interfaces, or else sometimes just a list of interfaces. It just so happens that 'sometimes' is the usual default case, but the alternate case can both be made to happen as well as just intermittently happen. The ftdi chips usually used with ice40's are always '2-interface' versions.

 I wonder if it would be similarly broken if someone tried to use a ft232h or so with just one interface?

benSTMax, 2019-01-09:
I get the same error using the Icestorm 0.4.0-dev and the iCE40-HX8K breakout board.
The driver is already using LibUsbK
Windows 10 version 1803 (Build 17134.407)
I have no other FTDI devices connected to my computer. The iCE40 board is the only one having an FTDI chip.

RGD2, 2019-01-09:
Win10 seems to make ftdi_usb_get_strings() behave differently somehow.
I think the win10 new behaviour could/will have to be worked around in icestudio’s code. 
This is a total showstopper for icestudio when it happens to people who can’t work around it by unplugging something else temporarily. 
A VM with win7 is not the answer. If you’re going to use a VM to get it working, then install Singularity (which wants Vagrant when on windows, and either VMware or virtualbox) and set a Linux container up to run the Linux version of icestudio. 
Vagrant and Singularity together handle auto install and update of a minimised Linux installation, so it would be more secure, smaller and faster than running another copy of windows within a VM. Plus - you won’t need to buy another OS license. 

But I think this route too is mostly unacceptable: icestudio’s extreme ease of setup (especially for new users) is compromised with all the extra steps. 

Might be best to advise new users on windows to abandon trying to run USB connected ice FPGA for the time being, if they hit it. 

Easiest answer for them if they really want to: install a Linux distro that’s nice to new users, like Ubuntu LTS or Mint. Then they can run the Linux version of icestudio and use the vastly more capable and reliable Linux USB driver system to avoid this issue entirely.  

The other way around is to get a raspberry PI and use it as a network server for the FPGA. (Thus also avoiding this issue).
They’re cheap enough that this makes actually quite a lot of sense. In this case this should work fine with the windows version of icestudio. 
There is some extra setup involved there too though - mostly in setting up the RasPI appropriately. But once done, the PI works this way just with network, power and FPGA plugged in - no need for KVM.

EN10, 2019-01-16:
Still getting the same error with v0.4.0-dev, basically icestudio is not usable on windows.

My current workaround is use a live iso of debian.

[Debian live iso](https://cdimage.debian.org/debian-cd/current-live/i386/iso-hybrid/?C=S;O=D)

memoriasIT, 2019-01-26:
I fixed this problem by using USB 3.0 instead of 2.0.
My OS is W10, I think that the cause of the problem is due to drivers not being saved correctly with 2.0.

profeBlanco, 2019-02-11:
help!!!

RGD2, 2019-02-11:
I think will work for sure: Running the linux version in a VM. If you're really stuck, and you really need it working today, and you don't have another windows machine you can try -- then what I'd recommend is using virtualbox and setting up an ubuntu vm, then installing icestudio within the VM. 
There's an option in the virtualbox device menu to 'attach' a usb device to the VM -> use that, then it should work.

As far as I know, this bug doesn't show up under any OS other than win10. You could use a win7 vm, but that would actually be a lot more time/trouble to set up. Win7, on the other hand, will take longer to install, and then it will want a valid key, and to 'activate'. Even if you go through all that, it will likely take longer to install as well.

For ubuntu you just download the 'ubuntu desktop LTS' .iso, and then open virtualbox, add a 'new machine' and tell it where the iso is. (probably allow at least 40GB max hdd space, but leave it set to 'dynamic' to grow as needed, it'll probably end up under 8GB, and allow around 4GB ram if you can, but stay in the 'green' range on the settings slider). 
Then you click 'start', and it'll ask for the iso -- there's a little button on the right which will open a dialog box to find the ubuntu iso you just downloaded. 
In a minute or two, you'll get a dialog asking if you want to try or install -- click install. 
You will probably want the 'minimal installation'. Let it 'erase disk and install ubuntu' (it's talking about that empty max 40GB virtual disk, don't worry!). Don't bother with encryption. 
Give yourself a short username, and don't forget what password you choose (you'll need it later). Probably best to check the  'log in automatically' option too. 

getting python 2.7 and xclip installed on ubuntu (or any debian-based distro) is just a `sudo apt install python2.7 xclip` in a terminal (ctrl-alt-t to open) It'll ask for that password you set when installing ubuntu, at least the first time you run any command with `sudo`. Then get the  latest icestudio-*-linux64.AppImage file, right click it, properties, permissions and check the 'execute' / 'allow executing file as program' checkbox.
After which double-clicking the appImage will open it, and you're away.
You can install the 'virtualbox guest utilities' if you want to make the icestudio window appear by itself (not inside the 'ubuntu desktop' window).

RGD2, 2019-02-11:
note after you plug in the board, you have to 'choose' it from the usb part of the virtualbox devices menu. (you can automate this in the settings).

Also, it'll be fairly slow to upload -- because Oracle charges for the license to run at usb2 speeds. But you can trial that part for free for a month if you like.
You enable it in the usb settings (usb2.0) - and It'll tell you you have to download and install the addon for it. After a reboot it'll be faster. 
No point turning on usb3, there arn't any ice boards around than can benefit from it.

I just ran through all this to make sure it all works (albeit under a linux host, there were actually a couple extra steps to enable usb) and the final 'vm hdd file' disk usage turned out to be 5.7GB.

profeBlanco, 2019-02-12:
Thank you, for your help, friend.
A thread brought me here. But I have not had problems with icestudio, but if I have this error in platformio running in visual studio when wanting to load the bitstream.

jpfersich, 2019-02-12:
It would be nice if this thread was on the github website. Though I always use Ubuntu’s Linux. 

/*—————————————————-*/
https://boincstats.com/signature/-1/user/51616339056/sig.png
See https://objectnets.net and https://objectnets.org

Obijuan, 2019-05-21:
We've solve a bug in Icestudio for uploading the bitstreams on windows. Please, update the toolchain (Tools/toolchain/Update), install the driver and try

benwynne, 2019-06-08:
Any updates here? I'm having the same problem. 

Using APIO 0.4.0b5

TimRudy, 2019-08-18:
I tested by updating toolchain. Icestudio shows Apio 0.4.0b5.

There is a fix, yes, I was able to use board on USB 2.0 port where I was not able to before. (*) See below.

I was trying to find the fixed line of code for this (inside libusb on GitHub, for example, or Apio project) and didn't find it.

The toolchain update picked up Apio v0.4.0b5 which is dated Sept. 2018 - confirmed by looking in:
~\.icestudio\.cache\default-apio:  contains apio-0.4.0b5.tar.gz

Did my Icestudio download the latest Apio, v0.4.0 (dated Nov 2018)? - or even a later tool somewhere in there? Icestudio itself is dated Sept 2018. I'm just curious what happened between Sept 2018 and May 2019?

For reference,
libusb 1.0.22 is latest official release of libusb from a long time ago, March 2018,
libusb 1.0.23-rc1 is the only thing that happened between Nov 2018 and May, namely Apr 2019

*I can conclude that this solved my problem with Win10, but I am still able to get a lot of this familiar error, on USB 1.0 (not supported, that's fine) and on USB 3.0:

> ~\.icestudio\apio\packages\tools-system\bin\lsftdi.exe
Number of FTDI devices found: 2
Checking device: 0
ftdi_usb_get_strings failed: -4 (libusb_open() failed)

1138-4EB, 2019-08-20:
@TimRudy, @RGD2, @EN10, regarding your comments about using alternatives such as VMs with Windows 7, a live ISO (on a USB drive), or a remote intermediate machine (say RPi), I have successfully programmed an icestick from a Docker Desktop (GNU/Linux) container on Windows 10. This allows to use the *"vastly more capable and reliable Linux USB driver system"*. 

The caveat, as explained in [cliffordwolf/icestorm#141](https://github.com/cliffordwolf/icestorm/issues/141#issuecomment-5228511449), is that Hyper-V VMs do not allow to share USB devices. Fortunately, USB/IP works. Details are available at https://github.com/ghdl/docker/tree/master/usbip. Furthermote, the same USB/IP client in the container can be used to program devices connected to a remote machine (i.e. RPi), if that is the case.

closed 2019-11-05 by Obijuan

Adrian0520, 2020-06-30:
Ami me pasa igual, en windows 10 me sale un error y la verdad no he podido solucionarlo, me he comprado la Alhambra 2 y no he podido hacer nada, estoy bastante desepcionado, ya que no encuentro una solución por ningún lado, por lo menos deberían subir un tutorial en youtube para mostrar una solución, segun ellos afirman que funciona en windows, lunux, y mac pero ahora veo que en windows 10 no me funciona, de momento creo que he perdído el dinero comprando la fpga alhambra 2

Obijuan, 2020-06-30:
Hola @Adrian0520,

Envía un mensaje a la lista de FPGAwars: [https://groups.google.com/g/fpga-wars-explorando-el-lado-libre](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre) indicando el error que te sale

Ahí  hay mucha gente que está usando Icestudio en Windows 10 con la Alhambra II y otras placas con FPGAs libres. En windows los drivers dan guerra la primera vez, que pero una vez que tienes claro  el proceso, funcionan bien

Este es la wiki con la información de puesta en marcha de la Alhambra II

[https://github.com/Obijuan/videoblog/wiki/Cap%C3%ADtulo-27:-FPGA:-La-placa-Alhambra-II#cargando-el-circuito-hola-mundo](https://github.com/Obijuan/videoblog/wiki/Cap%C3%ADtulo-27:-FPGA:-La-placa-Alhambra-II#cargando-el-circuito-hola-mundo)

La herramienta ha evolucionado mucho desde que se hizo ese tutorial, por lo que la apariencia de los pantallazos será un poco diferente. Te resumo los pasos:

* Instala la última versión (nightly built): Descárgala de aquí: [https://github.com/cavearr/icestudio-nightly/raw/master/0.5.1n200523/icestudio-0.5.1n200523-win64.exe](https://github.com/cavearr/icestudio-nightly/raw/master/0.5.1n200523/icestudio-0.5.1n200523-win64.exe)
* Las nuevas toolchain funcionan en python3, por lo que es importante que desinstales python2 antes de arrancar el instalador (te instalará python3, si no lo tienes ya)
* La instalación del driver es el paso más crítico, sigue las instrucciones que este tutorial, en el apartado de windows

https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/wiki/V%C3%ADdeo-2:-%C2%A1Poniendo-en-marcha-Icestudio!

En los talleres que damos en la Univesidad dedicamos media hora a su instalación y puesto en marcha. Cada ordenador con Windows es un mundo y hay que hacer varios ajustes y pruebas hasta que funcione.

Por eso es importante que escribas un mensaje al grupo indicando el error que te sale para que te podamos indicar pruebas a realizar para que te funcione bien

Este vídeo lo sacó hace unos días Russell Merrick donde muestra algunos ejemplos de uso de Icestudio en la placa bo-board en Windows. Comenta algunas incidencias que tuvo con los drivers, pero que puso subsanar. En su caso era debido a que tenía otros dispositivos que iban por el FTDI que usaban otro driver diferente. Y no conseguía instalador correctamente el de la alhambra II

[https://www.youtube.com/watch?v=pZWFmupcMTs](Russell Merrick)

Adrian0520, 2020-07-01:
> Hola @ Adrian0520 ,
> 
> Envía un mensaje a la lista de FPGAwars: https://groups.google.com/g/fpga-wars-explorando-el-lado-libre indicando el error que te sale
> 
> Ahí hay mucha gente que está usando Icestudio en Windows 10 con la Alhambra II y otras placas con FPGAs libres. En windows los drivers dan guerra la primera vez, que pero una vez que tienes claro el proceso, funcionan bien
> 
> Este es el wiki con la información de puesta en marcha de la Alhambra II
> 
> https://github.com/Obijuan/videoblog/wiki/Cap%C3%ADtulo-27:-FPGA:-La-placa-Alhambra-II#cargando-el-circuito-hola-mundo
> 
> La herramienta ha evolucionado mucho desde que se hizo ese tutorial, por lo que la apariencia de los pantallazos será un poco diferente. Te resumo los pasos:
> 
> * Instala la última versión (versión nocturna): Descárgala de aquí: https://github.com/cavearr/icestudio-nightly/raw/master/0.5.1n200523/icestudio-0.5.1n200523-win64.exe
> * Las nuevas herramientas funcionan en python3, por lo que es importante que desinstale python2 antes de organizar el instalador (te instalará python3, si no lo tienes ya)
> * La instalación del conductor es el paso más crítico, sigue las instrucciones que este tutorial, en el apartado de windows
> 
> https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/wiki/V%C3%ADdeo-2:-%C2%A1Poniendo-en-marcha-Icestudio !
> 
> En los talleres que damos en la Universidad dedicamos medios hora a su instalación y puesto en marcha. Cada computadora con Windows es un mundo y hay que hacer varios ajustes y pruebas hasta que funciona.
> 
> Por eso es importante que escribas un mensaje al grupo que indica el error que la venta para que podamos indicar pruebas a realizar para que funcione bien
> 
> Este video lo sacó hace unos días Russell Merrick donde muestra algunos ejemplos de uso de Icestudio en la placa bo-board en Windows. Comenta algunas incidencias que tuvo con los conductores, pero que puso subsanar. En su caso era debido a que tenía otros dispositivos que iban por el FTDI que usaban otro controlador diferente. Y no conseguimos instalador correctamente el de icestudio
> 
> https://www.youtube.com/watch?v=pZWFmupcMTs

Obijuan muchas gracias, a propósito soy alumno tuyo te sigo en youtube y vi todo el curso completo de fpgas, mira Obijuan te comento lo he solucionado de la forma mas fácil, me he descargado una maquina virtual (virtual box machíne), y he instalado en dicha maquina Manjaro KDE, una vez estando ejecutado manjaro en la máquina virtual me descargue el Icestudio y simplemente clic derecho en el archivo descargado, propiedades, permisos y activar la pestaña ejecutable, Obijuan lo explica muy bien en el video tutorial de como instalar Icestudio, y ya esta funcionando perfecto ni siquiera tube que instalar ningún otro programa ni controladores super fácil. Mil gracias Obijuan, here una gran persona, desde Colombia saludos

Adrian0520, 2020-07-01:
Estaría genial un tutorial de como diseñar el circuito de un variador de velocidad (VFD) para un motor ac trifasico, yo estoy en un proyecto de desarrollo de un dron híbrido de 200 hp de potencia, hemos elegido el motor de 1000 centímetros cúbico de la Yamaha Yzf R1

Obijuan, 2020-07-01:
Ok, me alegro de que lo tengas funcionando.
No obstante, no hace falta que uses la máquna virtual. Icestudio funciona de forma nativa en winwows 10. Te animo a que lo pruebes y a que nos des feedback de los problemas encontrados, para que esa información les pueda servir a otros.

Suerte con tu proyecto, tene una pinta muy buena 🙂

Adrian0520, 2020-07-01:
Muchas gracias Obijuan, a penas pueda te envío foto de esta aeronave que estamos construyendo, calculamos que va a quedar con una capacidad de carga de unos 200 kilos y una autonomía de vuelo de 4 horas, también te voy a mandar fotos de los motores eléctricos que estoy desarrollando, copie el diseño del motor del Tesla Model S es un motor de inducción trifasico con jaula de ardilla de cobre, este diseño de motor es libre, os voy a compartir todo el proyecto esperamos conseguir el récord mundial del dron quadcopter más potente del mundo, tengo dos opciones, comprar los variadores de velocidad (VFD) o diseñar el circuito con la FPGA, lo único es que aún me siento muy escaso de conosimiento para diseñar ese circuito, yo diría que nos vendría muy bien un tutorial de como hacer el circuito de un VFD y un ESC, ambos son muy parecidos el ESC es el que controla la velocidad en los motores brushless