FPGAwars/icestudio #242: libfuse error

mattvenn, 2018-06-01:
matt-pc:2020: ./icestudio-0.3.3-beta-linux32.AppImage 
dlopen(): error loading libfuse.so.2

```
AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/probonopd/AppImageKit/wiki/FUSE 
for more information
```

but libfuse is installed:

```
matt-pc:2022: ldconfig -v | grep fuse
	libfuse.so.2 -> libfuse.so.2.9.4
```

wd5gnr, 2018-08-30:
Same issue here. I can mount though but then it has problems finding its files:

./icestudio.AppImage --appimage-offset
sudo mount -o loop,offset=XXXX /mnt    # XXXX is the number from appimage-offset

mattvenn, 2018-09-03:
ubuntu 16.04 LTS

wd5gnr, 2018-09-05:
I am using KDE Neon which is based on Ubuntu

mattvenn, 2018-10-15:
I was trying the 32bit version on my 64bit machine! I just downloaded the 64bit version and it's working!
Sorry!