FPGAwars/icestudio #244: Cannot create virtualenv in CentOS 7

abdullahyildiz, 2018-06-04:
This is what I get when I run icestudio:

[image]

Do you have any idea about how I can fix that?

abdullahyildiz, 2018-06-04:
I also tried building directly from the source code. However, this time I get **Internet connection required** error.

rcardenes, 2018-10-10:
This is still an issue under the recently released v0.3.3. Shouldn't it be reopened?

Jesus89, 2018-10-12:
Hi,

The `--always-copy` flag was originally added because of an error in Windows. So I have updated the code to use the flag only for Windows: https://github.com/FPGAwars/icestudio/commit/3a7386ab48073cea48fdfcb46e8ec1b95e38be17 and avoid this issue in CentOS.

This will be included in the next release.