FPGAwars/icestudio #258: Problemas al cargar en la BX

ionhsFP, 2018-08-23:
Hola Jesus tengo problemas para cargar los bloques con el funcionamiento a la placa TinyFPGA BX. Como tenía el ordenador fatal desde la última actualización de W10 x64 y tampoco conseguia cargar a la placa lo he formateado todo. Así que tengo un W10 x64 pro recien formateado. He seguido los pasos de:

- Bajar he instalar icestudio 0.3.3 rc y python 2.7.13
- Abrir el icestudio y instalar el toolchain (sin problemas)
- Habilitar drivers, conectar la fpga hasta windows detecte la placa (com3)
- Se ha abierto la ventana y he elegido interface 0 
- Se ha abierto el zadig pero al darle a descargar ha dado un error (dejo captura)
- Le he dado a aceptar y quitar la placa y volver a ponerla.
- Seleccionado la placa tinyFPGA BX pero al cargar, no lo hace, pone cargando y al rato sale un cuadro rojo sin ningún texto.

Al volver al hacer el paso de instalar drivers cambia el mensaje de los pasos que hay que dar, así como no deja volver a seleccionnar el interface 0. Incluso los pasos son diferentes y el punto 3 parece que tiene alguna errata (dejo foto). Al pulsar OK carga una pantalla sobre "virtual serial driver" en la que no se puede hacer nada salvo pulsar done. (Dejo foto).

Hay algo que no está funcionando bien pero no consigo saber que es ni tengo claro que pruebas más se pueden hacer
[image]
[image]
[image]

ionhsFP, 2018-08-28:
Hola Jesus, Juan me ha comentado que a la Tiny hay que instalarle el bootloader y que por eso da error. Siento la molestia, puedes dar por cerrado este tema.

closed 2018-08-28 by ionhsFP

Jesus89, 2018-08-28:
Hola. De todas formas es un buen apunte, así que voy a plantear si puedo incluir la detección e instalación del bootloader desde Icestudio.