FPGAwars/icestudio #259: Programación en remoto de la tarjeta FPGA

fernandomosquera, 2018-08-24:
Vivado tiene una opción para programar una tarjeta en remoto mediante una conexión IP. 
Para ello, en el equipo con la tarjeta FPGA se  instala un pequeño programa Vivado Hardware Server (propietario de Xillinx) que abre un puerto y permite la conexión Ip con la tarjeta desde un equipo remoto.   
Para tarjetas FPGA Altera existe Altera Remote JTAG Server que es similar.
Aunque es algo antigua, en esta web explican los dos sistemas.  https://www.doulos.com/knowhow/fpga/remote_programming/

Podríais pensar en algo similar y añadirlo a IceStudio? 

Tendrías que crear un programa que actúe como servidor ejecutándose en  un equipo con la tarjeta FPGA conectada por USB  y en otro equipo usar IceStudio y que conecte mediante IP al servidor remotamente. 
Existe una solución libre en Github para tarjetas Xillinx pero usa JTAG llamada Xillinx Virtual Cable Server (XVC)  en https://github.com/Xilinx/XilinxVirtualCable
Y esta es la solucion adpatada para usar XVC en Rapsberry  en la cual se usan pines de sus salidas para conectar al JTAG de la tarjeta  https://github.com/derekmulcahy/xvcpi

Finalmente para trabajar en JTAG sobre puerto FTDI se puede usar Xilinx Virtual Cable Daemon   https://github.com/tmbinc/xvcd

El Xillinx Virtual Cable Server es libre (protocolo y especificaciones), pero claro depende del Vivado (que es propietario)
La idea seria similar, aunque open source en los 2 lados (server and client)

Gracias

Jesus89, 2018-08-24:
Buenas @fernandomosquera.

Esta funcionalidad ya está implementada en Icestudio. En la documentación puedes ver cómo configurar un host remoto por SSH: https://icestudio.readthedocs.io/en/latest/source/howto.html#configure-a-remote-host.

En el siguiente video (de 2016) se muestra un ejemplo de una Rapsberry Pi configurada como host remoto de Icestudio. Al ejecutar los comandos Verify, Build y Upload, se sincroniza el diseño en el remoto y se ejecutan los comandos en remoto:

[image]

Un saludo.

closed 2018-08-24 by Jesus89

1138-4EB, 2018-08-24:
Hola @Jesus89, ¿sería posible mantener esta issue abierta para valorar alternativas como las que comenta @fernandomosquera? La solución por SSH requiere disponer de un usuario y condece acceso a la máquina remota, no sólo a apio.

benitoss, 2018-08-24:
Yo me refiero a poder acceder directamente desde IceStudio a la tarjeta mediante una conexión IP como si estuviera conectada en local. Es una solución transparente al usuario.
Tanto las tarjetas de Xillix como de Altera disponen de ese sistema, como comento inicialmente.
El acceder mediante SSH a la maquina remota, seria el equivalente a acceder mediante VNC o remote desktop en Windows y ese no es el objetivo.