FPGAwars/icestudio #260: Crash al pegar múltiples veces

angelmicelti, 2018-08-25:
En la versión 0.3.3.rc, en Windows 10, al pegar un bloque (para duplicarlo) mediante CTRL-C / CTRL-V , la ventana del programa se cierra.
No sucede con los bloques de puertas lógicas pero sí, por ejemplo, con los FF tipo T y D, y me di cuenta al trabajar con un JK que había "fabricado" yo.

angelmicelti, 2018-08-28:
Creo que tiene que puede tener algo que ver con la imagen .svg que representa al bloque. He probado a "resetear  SVG" para dejar el bloque sin imagen y puedo copiar y pegar todas las veces que quiera el bloque resultante, es decir, ya no hay bug.

Jesus89, 2018-08-28:
Hola, puedes adjuntar los bloques que daban error en este issue?

angelmicelti, 2018-08-28:
Aquí te los dejo
**JK.ice** puede copiarse y pegarse con normalidad y no tiene imagen.
**JK2.ice** tiene un svg que bajé de Internet y **JK3.ice** tiene un svg que he creado yo con Inkscape. A ninguno deja copiarlo más de dos veces. A la tercera, crash.
**02-Biestable-D-0.ice** lo encontré en una de las colecciones que hay disponibles en GitHub (no recuerdo exactamente dónde). Con la imagen que trae, produce el mismo crash al pegar la tercera vez. Si reseteo el svg, obtengo **02-Biestable-D-0sin.ice**, que no da ningún fallo.
Por último, adjunto **FlipFlopJK.ice** y **FlipFlopJKsin.ice**: son los que yo "fabriqué". El primero tiene imagen, y da el fallo. El segundo, no tiene imagen y tampoco da fallo.

[Archivos.zip](https://github.com/FPGAwars/icestudio/files/2329303/Archivos.zip)

angelmicelti, 2018-08-28:
Sucede también con gran cantidad de archivos de diferente tipo. En cuanto se quita la imagen, el problema desaparece.

Jesus89, 2018-09-03:
Buenas @angelmicelti,

Creo que este bug puede tener relación con este otro: https://github.com/FPGAwars/icestudio/issues/196. Se trata de un bug en la versión de node.js implementada en NW.js 0.12.3, que es lo que utiliza actualmente Icestudio.

Muy probablemente se solucione en la próxima versión 0.4.0-beta, ya que reemplazaré NW.js 0.12.3 por Electron, que tiene una versión actualizada (y con el bug corregido) de node.js.

Un saludo.

CAMTEC, 2019-04-13:
Hola, espero me puedan ayudar, recién estoy aprendiendo a utilizar el icestudio, y tengo -o más bien tenía- el mismo problema que mencionan, a partir de incluir las colecciones de apoyo de Obijuan, noté que no podía copiar-pegar más de dos veces cualquiera de los elementos en las colecciones, por que icestudio se cierra al momento, no marca error, solo desaparece de pantalla.

No fue un gran problema, por que lo que hacía era que agregaba directamente de la colección los elementos que necesitaba sin utilizar el copiar-pegar.

Y digo que tenía ese problema, por que agregando más de las colecciones de Obijuan, en algún punto, el problema empeoró al punto que no puedo agregar más de dos o tres elementos del mismo, ya tipo ni siquiera de forma directa desde la colección, por que icestudio simplemente desaparece de pantalla.

Por lo que en principio no puedo generar un circuito con más de dos o tres elementos iguales, por que icestudio simplemente desaparece de pantalla.

Como prueba, desinstalé icestudio y lo volví a instalar, y el problema persiste de forma muy extraña, por ejemplo, con solo la colección por defecto, si agrego dos 0 o dos 1, al tercer elemento igual que agregue, icestudio se cierra al momento.

Si por ejemplo, agrego varios de los otros elementos de la colección por defecto, aparentemente, puedo agregar los que quiera  y no hay problema -o al menos yo no he notado que al llegar a cierto número de elementos agregados, aparezca el problema-, pero si agrego dos 0 o dos 1 seguidos (o una combinación de ellos), invariablemente, al tercer elemento;  icestudio desaparece de pantalla.

Uso Windows 10 y la version 0.4.0-dev, no tengo identificado otro problema que yo haya detectado con otro producto de software en mi equipo y puedo enviar toda las información de mi sistema como se requiera y/o hacer las pruebas que sean necesarias.

Espero, me puedan ayudar con este issue que se presenta con icestudio y mi sistema.

¡Saludos y gracias!

Obijuan, 2019-11-04:
Solucionado en la versión 0.5.1-nigthly
😀 (reportado por @angelmicelti) ¡Gracias!

closed 2019-11-04 by Obijuan