FPGAwars/icestudio #265: Las colecciones no se instalan en Mac OSX

xv5004, 2018-08-28:
Saludos.
Siguiendo los tutoriales llego al que instalamos las colecciones en IceStudio; sigo las instrucciones al pie de la letra pero no se me instalan.

Maquina: iMac
SO: macOS High Sierra
IceStudio Version 0.3.2

Pantallazos del proceso:
[image]
[image]
[image]

Lamento no poder proporcionar mas informacion, pero no me da ninguna; ni fallos, ni iformes de error ni nada. Sencillamente no ocurre nada.

Alguna idea?

Jesus89, 2018-09-03:
Hola,

Puedes probar con esta versión? https://github.com/FPGAwars/icestudio/releases/tag/v0.3.3-rc.

charliva, 2019-02-04:
I confirm this bug is solved and the latest version works correctly in macosx Mojave.

closed 2020-03-31 by cavearr