FPGAwars/icestudio #267: Low resolution on retina mac when opening a new window.

PhilippvK, 2018-08-29:
The UI looks fine and sharp on my high DPI MacBook screen but If I open a new instance of icestudio using `File -> New`, the new window look blurry (non retina optimized) like you can see here:

[image]

I think this happens on every MacOS version (my version is Mojave) but a fix has a very low priority.

I figured out, that in `app/scripts/services/utils.js`, where the new windows is beeing created, the command looks like this for me:
`/.../icestudio/node_modules/nw/nwjs/nwjs.app/Contents/Frameworks/nwjs\ Helper.app/Contents/MacOS/nwjs\ Helper /.../icestudio/app/`
Running this command in the Terminal has the same result. But if i use `nwjs.app/Contents/MacOS/nwjs` istead of `nwjs.app/Contents/Frameworks/nwjs\ Helper.app/Contents/MacOS/nwjs\ Helper` it is looking much better. I do not really know, what the difference between `nwjs` and `nwjs Helper` is and why `` results to this path, so I do not have any more ideas on how to fix it.

charliva, 2019-02-01:
Hi! I am working on this error, it is more important than it seems. In the next few days I will make an extraction request with the solution

Jesus89, 2019-02-02:
Closed via https://github.com/FPGAwars/icestudio/pull/303

closed 2019-02-02 by Jesus89