FPGAwars/icestudio #269: Crash when having three instances of the same objetc

andrespmt, 2018-09-03:
Windows 64-bits
0.3.3-RC

- Open Icestudio
- Include three AND gates 
- crash

I have been able to reproduce with AND, OR, MUX, etc.... cannot reproduce with Inputs / Outputs

andrespmt, 2018-09-03:
I went back to the default COLLECTION and it works.
So, my issue is when using Collection Academia_Jedi_22, 21, 20 ...
I think i read in another ticket that might be related to Components that include SVG ?

Went back to 0.3.0, 0.3.1, 0.3.2 , with collections 10,11 ... and i see the same issue, which is weird becasue i should have noticed before. Perhaps is something wrong in my computer and also when using Windows?

andrespmt, 2018-09-03:
Will be good if another person with Windows 64, can give a try.

Jesus89, 2018-09-03:
Duplicated: https://github.com/FPGAwars/icestudio/issues/260

closed 2019-11-05 by Obijuan