FPGAwars/icestudio #270: Sintetización errónea usando copia-pega en Windows

ximocat, 2018-09-06:
Haciendo el ejercicio 2 del tutorial 26 de electrónica digital de Obijuan me ha sucedido una cosa muy extraña.

1) (Windows 7, versión 0.3.3-rc) Construyo el circuito siguiente (copiando y pegando las tiras de timers y leds), lo guardo (ya que debido al bug #196 si no lo hago, al pulsar CRTL+U se cierra icestudio https://github.com/FPGAwars/icestudio/issues/196), lo sintetizo y el resultado no es el esperado: Únicamente se enciende el LED0 y no se propaga al resto de LEDs. El archivo con el circuito es el "Ejercicio26_2_malo.ice"

[image]

2) (Windows 10, versión 0.3.3-rc) Abro el archivo "Ejercicio26_2_malo.ice", lo sintetizo y da el mismo resultado: solo se enciende el led0 y no se propaga al resto

3) (Ubuntu 18.04, versión 0.3.3-rc) Abro el archivo "Ejercicio26_2_malo.ice", lo sintetizo y da el mismo resultado: solo se enciende el led0 y no se propaga al resto

4)  (Windows 7, versión 0.3.3-rc) Construyo el mismo circuito pero sin copiar y pegar y el resultado es el esperado: El encendido se propaga desde el LED0 al LED7. Archivo "Ejercicio26_2.ice"

Lo extraño es que a nivel de icestudio, los circuitos aparentemente son idénticos, pero el resultado al sintetizar es muy diferente

Gracias

ximocat, 2018-09-06:
En el siguiente zip van los dos archivos .ice, el que funciona y el que no (lleva el sufijo "malo")

[Tutorial 26_2.zip](https://github.com/FPGAwars/icestudio/files/2357030/Tutorial.26_2.zip)

Jesus89, 2018-09-16:
Buenas, he encontrado el problema. Los bloques timer estaban duplicados y superpuestos y las conexiones de entrada y salida estaban en distintos bloques. He detectado el bug viendo que el fichero "malo" tenía más líneas (más bloques) que el bueno.

Adjunto un GIF en el que se ven los bloques duplicados y cómo los reemplazo para obtener un circuito funcional.

[image]

Un saludo.

closed 2018-09-16 by Jesus89