FPGAwars/icestudio #276: Board dependent examples

esden, 2018-09-20:
Hi,

Unless I am missing something there is only one set of `File->Examples` menu files. These examples are available only for the Alhambra development board.

Is there a way to add .ice files for a different FPGA board?

What I am imagining is that the example projects in the File menu would change depending on the selected target development board. I would like to add a ported/adjusted set of examples for the iCEBreaker development board to the icestudio project. As far as I understand I would have to create a fork of the project and distribute my own icestudio modified to work with the iCEBreaker, I would rather point the users to your original project instead.

If I am missing something please let me know.

As always thanks for the amazing work! :)

esden, 2018-09-20:
For reference I have created a repository with "converted" examples. https://github.com/icebreaker-fpga/icebreaker-icestudio-examples Over time I would like to adapt those examples to be better suited to the iCEBreaker board. We don't have exactly the same amount of buttons and LED on the iCEBreaker so it will make sense to organize the examples little bit differently so the examples feel bit more "natural". :)

cavearr, 2020-01-29:
on v5.1.0 dev we improved migration between boards but for the moment it is not worked.

I labelet it as feature request because is not a bug, but it is interesting for the future.