FPGAwars/icestudio #277: Urgent simulation for icestudio!

alico797, 2018-09-23:
I would like that icestudio had the option of simulation, of the logical states, displays, etc. It would be of great help, especially for those who do not yet have an FPGA board. From my point of view, I think it would be a breakthrough for the program...

harevalop, 2019-07-02:
> Me gustaría que icestudio tuviera la opción de simulación, de estados lógicos, displays de 7 segmentos, etc. Sería de gran ayuda, especialmente para aquellos que aún no tienen una placa FPGA. Desde mi punto de vista, creo que sería un gran avance para el programa ...

Quizá este link pueda ser de ayuda:
https://www.bluetin.io/displays/simulate-oled-lcd-display-pc-pil-opencv/

qpeters, 2021-07-13:
@alico797 since this feature can take long to be implemented you might want to try [hneemann/Digital](https://github.com/hneemann/Digital).

> Components can be described using VHDL or Verilog. The open source VHDL simulator ghdl needs to be installed to simulate a VHDL defined component, and the open source Verilog simulator Icarus Verilog is required to simulate a Verilog defined component.