FPGAwars/icestudio #284: Long menus not visible

drtrigon, 2018-10-28:
If I use https://github.com/set-soft/collection-kefir_i the menus do sometimes get so long (e.g. Varios > Buses > Unir > Join_wires) that some items cannot be selected. As the menu does not react to keyboard actions I see no way to select the items at the end of the list.

drtrigon, 2018-11-02:
Do you need more information on this? Does a work-a-round exist?

TimRudy, 2018-12-16:
I'm working on scrolling for long menus, fix coming soon

TimRudy, 2019-07-09:
I explored a fix using bootstrap class="pre-scrollable" and by trying to override bootstrap menu CSS properties. I could not find a way. These broke the menus and were incompatible with submenu display. The menus (which are pretty nice and elegant) would have to be rewritten completely using something beyond bootstrap, I think.

The hint that menus do not react to keyboard actions (up, down arrow keys) is a good idea for a workaround fix. Along with that, however, there should be dynamic repositioning/scrolling so that the parts that are hidden become visible!

drtrigon, 2019-07-10:
Dynamic repositioning would be great. Alternatively show at least the name of the selected part in e.g. the bottom status bar.

Obijuan, 2022-05-02:
Collections are now displayed in a different panel, and the components are no longer accesible through menues.
I will close this issue. Feel free to reopen it if the error remains

closed 2022-05-02 by Obijuan