FPGAwars/icestudio #285: Getting error "unknown model '$_DFFSR_PPP_'" when using "Add a block..."

marcio-ao, 2018-11-09:
Hello,

First of all, thank you so much for IceStudio. This is an *outstanding* program! It is doubtful I would have ever gotten a change to try Verilog programming if it wasn't for this incredibly easy-to-use tool!

So anyhow, I built some code for sending different colors to an AdaFruit NeoPixel LED strip (based on the WS2812 RGBW LEDs).

I was able to make it work as one large Verilog code block, but then I found out about the "Add as Block..." function and decided to split my design up into different IceStudio files so that I could have a reusable component.

This is where I am running into problems. When I try using that block from another file, I am getting an `unknown model '$_DFFSR_PPP_'` error when building. Interestingly, simply changing one line to a literal rather than a load from memory causes that error to go away.

I have included the files that demonstrate the error. Can anyone help me figure out what is going on? The main file is "NeoPixel-Error.ice" and the block it includes is "neopixel_ws2812b_grbw.ice". Both are in the archive.

[design-with-error.zip](https://github.com/FPGAwars/icestudio/files/2567542/design-with-error.zip)

marcio-ao, 2018-11-09:
I guess at this point I'm not sure if this is an error with IceStudio, or the underlying toolchain. I tried grepping for "$_DFFSR_PPP_" in the exported Verilog file and there was nothing there.

daveshah1, 2018-11-09:
The problem is this line
```
data <= grbw;
```
Inside the asynchronous reset in v0b22ca_ve8cefb

Because the reset data is not to be constant, being the output of the memory, this would require a flipflop that can be asynchronously set or reset at runtime. iCE40 flipflops can either be set or reset but this must be selected at build time (there are theoretical workarounds for this involving two flipflops and latch to remember whether S or R was used but they would introduce more problems than they would solve). 

Changing this line to reset to a constant value instead should fix the issue.

Jesus89, 2018-11-09:
Thanks @daveshah1 for solving this one :wink:

marcio-ao, 2018-11-13:
Yes, thank you! BTW, I got it to work by following the advice. If anyone wants to see the improved block for driving NeoPixels, it is in [this repo](https://code.alephobjects.com/diffusion/CHIPLIB).