FPGAwars/icestudio #290: Error: angular is not installed.

bogdanr, 2018-12-13:
When I run `npm start` I receive this error:
```
nimblex:/mnt/sda1/icestudio# npm start

> icestudio@0.4.0-dev start /mnt/sda1/icestudio
> grunt serve

Running "nggettext_compile:all" (nggettext_compile) task

Running "watch:scripts" (watch) task
Waiting...
Running "wiredep:task" (wiredep) task
Warning: Error: angular is not installed. Try running `bower install` or remove the component from your bower.json file. Use --force to continue.

Aborted due to warnings.
Completed in 1.252s at Thu Dec 13 2018 21:05:50 GMT+0200 (EET) - Waiting...
```

This is my npm list:
```
nimblex:/mnt/sda1/icestudio# npm list
icestudio@0.4.0-dev /mnt/sda1/icestudio
├── angular@1.7.5
├── bower@1.8.4
├─┬ extract-zip@1.6.7
│ ├─┬ concat-stream@1.6.2
│ │ ├── buffer-from@1.1.1
│ │ ├── inherits@2.0.3
│ │ ├─┬ readable-stream@2.3.6
│ │ │ ├── core-util-is@1.0.2
│ │ │ ├── inherits@2.0.3 deduped
│ │ │ ├── isarray@1.0.0
│ │ │ ├── process-nextick-args@2.0.0
│ │ │ ├── safe-buffer@5.1.2
│ │ │ ├─┬ string_decoder@1.1.1
│ │ │ │ └── safe-buffer@5.1.2 deduped
│ │ │ └── util-deprecate@1.0.2
│ │ └── typedarray@0.0.6
│ ├─┬ debug@2.6.9
│ │ └── ms@2.0.0
│ ├─┬ mkdirp@0.5.1
│ │ └── minimist@0.0.8
│ └─┬ yauzl@2.4.1
│   └─┬ fd-slicer@1.0.1
│     └── pend@1.2.0
├─┬ UNMET PEER DEPENDENCY grunt@1.0.3
│ ├── coffeescript@1.10.0
│ ├─┬ dateformat@1.0.12
│ │ ├── get-stdin@4.0.1
│ │ └─┬ meow@3.7.0
│ │   ├─┬ camelcase-keys@2.1.0
│ │   │ ├── camelcase@2.1.1
│ │   │ └── map-obj@1.0.1 deduped
│ │   ├── decamelize@1.2.0 deduped
│ │   ├─┬ loud-rejection@1.6.0
│ │   │ ├─┬ currently-unhandled@0.4.1
│ │   │ │ └── array-find-index@1.0.2
│ │   │ └── signal-exit@3.0.2
│ │   ├── map-obj@1.0.1
│ │   ├── minimist@1.2.0
│ │   ├─┬ normalize-package-data@2.4.0
│ │   │ ├── hosted-git-info@2.7.1
│ │   │ ├─┬ is-builtin-module@1.0.0
│ │   │ │ └── builtin-modules@1.1.1
│ │   │ ├── semver@5.6.0 deduped
│ │   │ └─┬ validate-npm-package-license@3.0.4
│ │   │   ├─┬ spdx-correct@3.1.0
│ │   │   │ ├── spdx-expression-parse@3.0.0 deduped
│ │   │   │ └── spdx-license-ids@3.0.2
│ │   │   └─┬ spdx-expression-parse@3.0.0
[... 1445 lines cut ...]
  ├── file-exists@2.0.0
  ├── merge@1.2.1
  ├─┬ multimeter@0.1.1
  │ └── charm@0.1.2
  ├── rimraf@2.6.2 deduped
  ├── semver@5.6.0
  └─┬ yargs@3.10.0
    ├── camelcase@1.2.1
    ├─┬ cliui@2.1.0
    │ ├─┬ center-align@0.1.3
    │ │ ├─┬ align-text@0.1.4
    │ │ │ ├─┬ kind-of@3.2.2
    │ │ │ │ └── is-buffer@1.1.6 deduped
    │ │ │ ├── longest@1.0.1
    │ │ │ └── repeat-string@1.6.1
    │ │ └── lazy-cache@1.0.4
    │ ├─┬ right-align@0.1.3
    │ │ └── align-text@0.1.4 deduped
    │ └── wordwrap@0.0.2 deduped
    ├── decamelize@1.2.0
    └── window-size@0.1.0

npm ERR! peer dep missing: grunt@~0.4.0, required by grunt-wget@0.1.3
```
I have npm 6.4.1 and node 8.12

charliva, 2019-01-13:
Instala la última versión de node  y ejecuta de nuevo npm install (asegurándote que la nueva verisón de node ha sustituido a la actual).

bogdanr, 2019-01-14:
As you can see in the bug report I did that. The first package listed by `npm list` is angular.

charliva, 2019-01-14:
Sorry for my first response in Spanish :)

The error is a unmet peer dependency (UNMET PEER DEPENDENCY grunt@1.0.3) , this type of conflicts are usually due to the use of different node versions to which the project was developed.

I'll recomend you the next steps:

1) remove node_modules directory inside icestudio  or remove icestudio dir and clone from github again.
2) Install latest node.js version https://nodejs.org/dist/v11.6.0/node-v11.6.0-linux-x64.tar.xz and modify PATH environment variable to use it.
3) From your icestudio clean dir, execute "npm install"
4) After installation succed, execute "npm start"

I hope this help you

charliva, 2019-01-14:
Try to execute:

`npm install extract-zip --save`

and then:

`npm start`

bogdanr, 2019-01-14:
Now it looks exactly like the first error message.

charliva, 2019-01-14:
The problem resides in your local installation, probably  old packages override the new ones. Try next insice icestudio dir:

`npm cache clean
 npm update -g
npm install
npm start`

bogdanr, 2019-01-14:
This doesn't really do anything because I had a clean install.

charliva, 2019-01-14:
can you describe the steps you follow from scratch and the exact commands?
which linux distribution or mac do you have?

bogdanr, 2019-01-15:
I am using my own distro and building nodejs from source.
Yesterday when we did all the test I deleted all node_modules and related things so it would all be in a clean environment.

I am not doing anything fancy, just:
```
    1  git clone git@github.com:FPGAwars/icestudio.git
    2  cd icestudio/
    3  node --version
    4  npm install
    5  npm start
```
Obviously, after I did this I also did a `npm install extract-zip --save` because extract-zip was reported missing and then I tried another `npm start` which ended up looking like the first error message.

charliva, 2019-01-16:
Since you do not work on a standard distribution, I do not know how to help you.

Try to do something, sometimes with similar dependency problems like this, solved it:

`npm install -g npm-install-missing`

and then insice icestudio dir:

`npm-install-missing`

If previously doesn't work, remove icestudio dir , download this script and execute:

https://gist.github.com/charliva/f47e059425a375abb067e905d2caa8b1#file-icestudio-install-with-report-sh

this install icestudio and try to launch it. GEnerating a install-report.txt file inside the icestudio directory.

When finished  attach your report and i try to find the problem.

bogdanr, 2019-01-19:
[install-report.txt](https://github.com/FPGAwars/icestudio/files/2775337/install-report.txt)

cavearr, 2020-01-29:
you can try the new 0.5.1 dev branch with node 10.17.x installed.

closed 2020-01-29 by cavearr