FPGAwars/icestudio #292: Python 2.7 is not installed automatically, contrary to docs

gojimmypi, 2018-12-23:
The docs [here](https://icestudio.readthedocs.io/en/latest/source/installation.html#windows) state that Python will be installed automatically.

[image]

Upon launching, click on the "click here"...
[image]

Then on the "Python 2.7 not installed" message...
[image]
nothing happens.

Note that I already have Python 3.6 installed:
[image]

ximocat, 2018-12-27:
https://groups.google.com/d/msg/fpga-wars-explorando-el-lado-libre/wDVYpd6qNCo/VeCjulUtDQAJ

Obijuan, 2019-11-05:
Fixed In icestudio 0.5.1-nigthly. It has been migrated to python 3.7
You can download from here: https://icestudio.io/#lk-download

closed 2019-11-05 by Obijuan