FPGAwars/icestudio #293: Windows Installer wants Python 2.7.13 despite 2.7.14 already installed

gojimmypi, 2018-12-23:
As noted in issue #292 Python was not installed automatically from the [zip file](https://github.com/FPGAwars/icestudio/releases/download/v0.4.0-dev/icestudio-0.4.0-dev-win64.zip) version of IceStudio, nor does it detect if Python manually installed. I have Python 2.7.14 installed, and this time I am running the [downloaded full exe](https://github.com/FPGAwars/icestudio/releases/download/v0.4.0-dev/icestudio-0.4.0-dev-win64.exe) to install IceStudio.

[image]

I was prompted to install Python 2.7.13:
[image]

Clicking "yes" IceStudio install seems to have completed successfully, although Python 2.7.14 is still installed (not replaced with this installer)
[image]

I receive the same errors about the toolchain not installed, however when installing via the exe ... seems to have successfully installed the toolchain upon clicking the box. 

I'm using Windows 10 version 1809

ximocat, 2018-12-27:
https://groups.google.com/d/msg/fpga-wars-explorando-el-lado-libre/wDVYpd6qNCo/VeCjulUtDQAJ

Obijuan, 2019-11-05:
Please, try again with Icestudio 0.5.1-nigthly

closed 2019-11-05 by Obijuan