FPGAwars/icestudio #299: Error render html Data Output labels

charliva, 2019-01-28:
I'm publishing this bug because it is very annoying, since it happens whenever a design that uses the outputs D0-D7 is opened.

I am configuring an 8-bit output, assigning it D0-D7 alhambra-ii outputs.

When I am loading the .ice project, icestudio generates the tags badly (the internal output allocation is correct, but the html tags do not). Attached image as it loads the project file:

! [bug_labels_0] (https://user-images.githubusercontent.com/6904865/51859582-c5c97b00-2337-11e9-9524-1667aa66dd2a.jpg)

I'm reviewing the code, it's the first time I check it and I do not know it enough (in the time I spent I could not find the place where you do the label assignment), but I think the problem is because you use a regular expression to assign the html tag and use an incorrect regular expression or something like ** "indexOf" **, so if you check all the pins on the board, take the last correspondence so that it takes DD0 instead of D0, because in the file pinout.json for the board of the alhambra-ii DD0 is after D0.

If the order of the pins with conflicting labels is changed, everything works correctly because in this way D0 is after DD0 and the match is more consistent. I am attaching an image using modified pinout.json (also attached):

! [bug_labels_1] (https://user-images.githubusercontent.com/6904865/51859786-45574a00-2338-11e9-96fe-24e668b74831.jpg)

If someone suffers this error and wants to solve it before it is solved officially, you should only replace the attached pinout.json file, in the path icestudio / app / resources / boards / alhambra-ii / pinout.json

This error affects all tables with overlapping pin names (for example, DD0, D0) and in the inappropriate order in the json file.

I hope this helps you solve this error.

[pinout.zip](https://github.com/FPGAwars/icestudio/files/2804496/pinout.zip)

Obijuan, 2019-02-01:
I confirm this bug. Is the same in the with Alhambra I board

Jesus89, 2019-02-02:
Solved in the Alhambra II board (https://github.com/FPGAwars/icestudio/pull/300). Maybe we should apply the same script for the Alhambra I pinout.

Obijuan, 2019-02-02:
Done! I will close this issue

closed 2019-02-02 by Obijuan