FPGAwars/icestudio #305: Blank error messages in code block

charliva, 2019-02-08:
If you have some kind of syntax errors inside a code block, sometimes error message is blank and you don't know where is the problem.

At this moment, i 'm using output command viewer and open the temporaly main.v generated file to look for error.

I'm attaching icestudio with test design open (attach it too) with no command at the end of line.

An screenshot with blank error.

[image]

[tests_errores.zip](https://github.com/FPGAwars/icestudio/files/2844625/tests_errores.zip)

charliva, 2019-02-08:
I'm fixed it and I do a pull request.

An screenshot with problem solved:

[image]

Obijuan, 2019-02-08:
Checked! It works! :-)  Thanks!

closed 2019-02-08 by Obijuan

Jesus89, 2019-02-08:
Thanks @charliva!

TimRudy, 2019-12-23:
Can you check this on 0.5.1-nightly? iverilog or yosys errors are not piped to the existing command output window. It remains blank. However, if you open up a new output window then it shows the previous output.