FPGAwars/icestudio #309: Build only Linux executable

BryceBeagle, 2019-02-10:
I'm trying to turn this project into a package for Arch Linux.

Is there a way to build _only_ the Linux executable (not the AppImage) from source? Trying to run `npm run dist` gives me an error about not having Wine installed. 

As I don't need a Windows build, I don't think this should be necessary.

umarcor, 2019-12-25:
@BryceBeagle, you can patch `gruntfile.js` to overwrite `platforms` and `distCommands` just after:

https://github.com/FPGAwars/icestudio/blob/dcbbbb7f9e28fd74b129a3d1ac67e85a3886d58a/gruntfile.js#L16-L20

For example:

```js
    platforms = ['linux32', 'linux64'];
    distCommands = ['compress:linux32', 'compress:linux64'];
```

cavearr, 2020-01-29:
hi @BryceBeagle ! it isn't a linux executable as is.  Icestudio uses nwjs. If you want to package icestudio for specific linux flavor, you can find some information at nwjs:

http://docs.nwjs.io/en/latest/For%20Users/Package%20and%20Distribute/#linux

Thanks!

closed 2020-01-29 by cavearr